intel/libxcam #174: Completion of error handling

elfring, 2015-09-06:
Would you like to add more error handling for return values from functions like the following?
- [printf](http://pubs.opengroup.org/onlinepubs/9699919799/functions/printf.html "Print formatted output.") ⇒ [CLTnrImageKernel::print_image_histogram](https://github.com/01org/libxcam/blob/4a386ef4a67acab40e27f2b086f7a46a715f90fb/xcore/cl_tnr_handler.cpp#L597)
- [pthread_mutex_init](http://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_mutex_init.html "Initialise a mutex.") ⇒ [Mutex constructor](https://github.com/01org/libxcam/blob/6c9241d17dacf18147794f6eef03a61c7407c043/xcore/xcam_mutex.h#L36)

liuyinhx, 2015-10-08:
fixed.

closed 2015-10-08 by liuyinhx

elfring, 2015-10-08:
Are there [a few update candidates](https://github.com/01org/libxcam/blob/6c9241d17dacf18147794f6eef03a61c7407c043/xcore/xcam_cpf_reader.c#L85 "More update candidates") left over?