intel/libxcam #179: [xcam][master][CL] Preview failure when open SNR

liuyinhx, 2015-09-10:
Title:
[xcam][0.2.1/master/0.3][CL] Preview failure when open SNR

Contents:
BSP: Yocto-Linux
Version: Libxcam 0.2.1
HW: rubypark
Regression: new issue

Reproduce steps:
1. deploy the image to rubypark device
2. open SNR in config files
3. run command:
    test-device-manager -m dma -c -f BA10 -d still -p -a dynamic --enable-snr

Expect and current result:
Open SNR, the preview should be normal.

Bug description:
The preview will be stuck when open SNR, and the preview has only one frame.

liuyinhx, 2015-09-10:
This issue is similar to https://github.com/01org/libxcam/issues/154

windyuan, 2015-09-17:
fixed.
commit fb73ad7beec2968a33730da829de33a4ddf4d7d9

liuyinhx, 2015-09-17:
Verified, close this issue.

closed 2015-09-17 by liuyinhx