intel/libxcam #216: file handle leaks increase fast

windyuan, 2015-11-03:
when running xcamsrc,  there are lots file handle leaks, and number is increasing very fast every second. 
libxcam commit e965dff1305af665d13f84734227a5b484f61ed1
Steps:
1.  run
    $ gst-launch-1.0 xcamsrc io-mode=4 imageprocessor=1 analyzer=2 pipe-profile=0 ! video/x-raw, format=NV12, width=1920, height=1080, framerate=25/1 ! queue ! vaapiencode_h264 rate-control=cbr ! fakesink
2. get process_id and check file handles
   $ ps | grep gst-launch   # to get PID
   $ ls -l /proc/PID/fd | wc -l

Expect result:
   file handle(fd) keep the same number in running time.

Actual result:
   number of fd increasing fast

windyuan, 2015-11-03:
MengJia, please check this issue.
Thanks.

dspmeng, 2015-11-04:
commit f0f23a8b4f5ff14d601e4a4b0807fa10c139af3a drm: enable prime buffer sharing for cl buffer and image

With above commit, two hidden issue are triggered by our use case:
V4l2 exports dma buf fd -> DrmBoData is created and get its own fd via drm_intel_bo_gem_export_to_prime() -> drm prime driver find an already exported dma buf (by v4l2), and go to a specific path.

1. RBP Linux kernel 3.10.28 has a bug inside drm_prime.c, fixed later in [3.11] (https://github.com/dspmeng/linux/commit/4a88f73f14f6d6c94616538427e1235a6d0a5885). With this bug, driver already allocates fd, but failed to return it to user space. so drm_intel_bo_gem_export_to_prime() always failed in our user case, and has no way to close the fd.

2. DrmBoBuffer is [created] (https://github.com/01org/libxcam/blob/master/xcore/cl_image_processor.cpp#L116) for every incoming V4l2Buffer, and allocates a fd. But it didn't close the fd in destructor.

dspmeng, 2015-11-05:
Fixed by https://github.com/01org/libxcam/pull/217 and another linux kernel patch.

closed 2015-11-05 by dspmeng