intel/libxcam #220: xcamsrc hang when ctl-c quit with raw-file-input

windyuan, 2015-11-06:
ENV:
Linux
libxcam: commit 72d03bce350dec4306ae1669fd8e9f78f7a14948

Reproduce steps:
1. capture a raw video file
          e.g ./test-device-manager -f BA10 -m mmap -d still -s -n 10 # save a video stream on every 10th frame
2. simulate raw-file-input with xcamsrc plugin. 
      gst-launch-1.0 xcamsrc io-mode=4 imageprocessor=1 analyzer=2 pipe-profile=2 fake-input="capture_buffer.raw" ! video/x-raw, format=NV12, width=1920, height=1080, framerate=25/1 ! queue ! vaapiencode_h264 rate-control=cbr ! fake

3. CTRL+C to break gst pipeline

Expect Result:
    program quit soon.

Actual Result:
    sometimes, program hang there with dead-locks.  looks like FakePollThread hang there waiting for 
buffer release.

GDB  stacks.
Thread 1.
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../nptl/sysdeps/unix/sysv/linux/i386/i586/../i486/pthread_cond_wait.S:188
188     in ../nptl/sysdeps/unix/sysv/linux/i386/i586/../i486/pthread_cond_wait.S
(gdb) bt
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../nptl/sysdeps/unix/sysv/linux/i386/i586/../i486/pthread_cond_wait.S:188
#1  0xb72eddf5 in XCam::Cond::wait (this=0x87dfc58, mutex=...) at ../../xcore/xcam_mutex.h:79
#2  0xb723506b in XCam::Thread::stop (this=0x87dfc30) at xcam_thread.cpp:116
#3  0xb721fcb5 in XCam::PollThread::stop (this=0x89e8790) at poll_thread.cpp:161
#4  0xb7213b71 in XCam::DeviceManager::stop (this=0x874c750) at device_manager.cpp:274
#5  0xb72e67db in gst_xcam_src_stop (src=0x874e0e8) at gstxcamsrc.cpp:858
#6  0xb6fcaa4d in gst_base_src_stop (basesrc=basesrc@entry=0x874e0e8) at gstbasesrc.c:3418
#7  0xb6fd2b64 in gst_base_src_activate_push (pad=0x8750008, active=0, parent=0x874e0e8) at gstbasesrc.c:3571
....

 Thread 3
#3  0xb72102f0 in XCam::BufferPool::get_buffer (this=0xb29008e0, self=...) at buffer_pool.cpp:198
#4  0xb721966a in XCam::FakePollThread::poll_buffer_loop (this=0x89e8790) at fake_poll_thread.cpp:83
#5  0xb722062c in XCam::CapturePollThread::loop (this=0x87dfc30) at poll_thread.cpp:69
#6  0xb7234e5b in XCam::Thread::thread_func (user_data=0x87dfc30) at xcam_thread.cpp:67

windyuan, 2015-11-06:
If hard to reproduce, a easy way just modify kernel_bayer_pipe.cl and make OCL compiling failure. the program then failed to quit.

dspmeng, 2015-11-16:
Fix by https://github.com/01org/libxcam/pull/228

closed 2015-11-16 by dspmeng