intel/libxcam #229: [xcam][master][CL] stats info Y/RGB histogram calculation error

zongwave, 2015-11-16:
Almost half of all pixels were filled into histgram bin 0 in all light conditions.

debug trace of histogram y:
3960    0    0   16   43   82  100  166  164  198  194  144  193  217  295  324
 351  224  183  142  114   80   65   30   34   25   25   27   26   21   29   24
  16   15    7    5    7    6    5    7   10   11    8   10    8    9    5   11
   6    9   14    1    2   10    4    8    6    5    6    2    2    3    3    4
   3    6    5    1    1    6    7    4    5    6    5    8    4    3    5    6
   8    8    6    6    8    2    1    2    1    0    1    0    0    1    0    0
   1    1    4    0    0    1    2    0    0    1    1    0    0    0    1    0
   0    0    0    0    0    0    0    0    2    0    0    0    1    0    1    0
   0    0    0    0    0    0    0    0    0    1    1    0    0    0    1    0
   0    0    0    0    0    0    1    0    1    1    0    2    0    0    0    1
   0    0    0    0    1    0    0    0    0    0    1    0    1    0    1    0
   1    0    1    0    0    1    0    1    0    0    0    0    0    0    0    1
   1    1    0    0    0    0    0    0    0    1    0    0    0    1    0    0
   0    0    1    0    3    0    0    1    0    0    0    1    0    0    0    0
   0    0    0    0    1    0    0    0    0    0    0    1    1    0    2    2
   0    0    0    0    2    0    0    0    0    0    0    0    0    1    3  143

liuyinhx, 2015-11-18:
Update:
There has error in out_index calculation,  out_index has duplicate values:
out_index: 0
out_index: 0
out_index: 120
out_index: 120
out_index: 240
out_index: 240
out_index: 360
out_index: 360
out_index: 480
out_index: 480
out_index: 600
out_index: 600
......

windyuan, 2015-11-18:
good. looks like index calculation issue. Try on following patch

diff --git a/cl_kernel/kernel_bayer_pipe.cl b/cl_kernel/kernel_bayer_pipe.cl
index eefbaea..7363f1f 100644
--- a/cl_kernel/kernel_bayer_pipe.cl
+++ b/cl_kernel/kernel_bayer_pipe.cl
@@ -380,7 +380,7 @@ inline void stats_3a_calculate (

     if (l_id_x % STATS_3A_GRID_SIZE == 0 && l_id_y % STATS_3A_GRID_SIZE == 0) {
         float4 tmp_data;
-        int out_index = mad24(g_id_y / STATS_3A_GRID_SIZE,  g_size_x / STATS_3A_GRID_SIZE, g_id_x / STATS_3A_GRID_SIZE);
+        int out_index = mad24(get_group_id(1),  g_size_x / STATS_3A_GRID_SIZE, g_id_x / STATS_3A_GRID_SIZE);
         tmp_data = input[shared_pos (l_id_x + SLM_CELL_X_OFFSET, l_id_y + SLM_CELL_Y_OFFSET)];
         stats_output[out_index].avg_gr = convert_uchar_sat(tmp_data.x * 255.0f);
         stats_output[out_index].avg_r = convert_uchar_sat(tmp_data.y * 255.0f);

liuyinhx, 2015-11-18:
Fixed.

windyuan, 2015-11-18:
pull request https://github.com/01org/libxcam/pull/230

windyuan, 2015-11-19:
Hi Yinhang,
    please verify and close this bug.
Thanks,
Wind

liuyinhx, 2015-11-19:
Verified, close this issue.

closed 2015-11-19 by liuyinhx