intel/libxcam #337: 关于libxcam编译的问题

cherixy, 2016-07-14:
您好，打扰了。
看到你在维护持续维护libxcam，想向你请教一个关于libxcam编译的问题。
我使用 ./configure --enable-aiq --enable-libcl该命令，想启用aiq和cl功能，但是编译后生成的libxcam_core.a的静态库中并没有aiq和cl函数符号，也就是说 --enable-aiq 和--enable-libcl并没有使得对应的功能生效，不太清楚是否缺少了什么配置
希望得到您的指点，谢谢~~

zjamy, 2016-07-15:
Thank you for your follow on libxcam.

Please refer to https://github.com/01org/libxcam/wiki/Build. From this page, we can see "If --enable-aiq, need get ia_imaging lib which we don't support."
The reason is that, most of the aiq lib is close sourced. We supported some close sourced aiq lib. But we can't open out due to IP issues.

But would you please post your condition with which libxcam can help? Then we might figure out a way to help.

windyuan, 2016-07-15:
just tried, looks like we have a compile issue in recent framework change. will fix it first and check this issue.
Thanks.

cherixy, 2016-07-16:
@zjamy   Use xcam for the ISP-Control of camera.  I have tried to complie xcam with the tips of  https://github.com/01org/libxcam/wiki/Build .  However, it also show  'checking for LIBCL... no',when configure the xcam

zjamy, 2016-07-18:
now, do you passed the AIQ issue? If so, good. Let's move on to the CL part.
Now for CL part, we are using beignet(https://www.freedesktop.org/wiki/Software/Beignet/) to support openCL. Would you please share your HW plaftform, then I can have a check whether beignet could support you?

windyuan, 2016-07-18:
@cherixy , we just fixed a build issue in libxcam to make env clean. you can take a try. ISP and CL modules are separate from xcore, you can find them from modules/isp/ and modules/ocl/.
As @zjamy said, OCL driver, currently need you to compile and install beignet first. In next couple releases, we'll consider to remove this dependency if there's requirement of different ocl drivers.

cherixy, 2016-07-18:
@zjamy  I have solved the AIQ and CL issue. About the CL part,  it's ok after changing a PC with intel Display Card. Maybe like you said, beignet could't support the older one. Thank you very much~~
@windyuan  Currently, OCL driver is a little trouble for most users. Removing this dependency may be good.  Of course, Thank you too much.   
I don't know whether others concern to this project. I have libxcam-0.2.1,which comes from my colleague. I don't know where it comes from. Is it from yours?

windyuan, 2016-07-18:
@cherixy if you don't need ocl features, you can --disable-libcl. But we have lots features(Wavelet-NR, TNR, WDR, defog, smart analysis) developed and developing by OCL which are platform independent. If you have interests, you can check on wiki/tests.

yes, there's only one libxcam, it should be come from us. if you need get more detail. you can email me(feng.yuan@intel.com) or @zjamy (juan.j.zhao@intel.com)

windyuan, 2016-07-22:
@cherixy 
   if no more issues here, can we close this bug?
Thanks
Wind

zjamy, 2016-07-22:
@cherixy Do you have some modifications on this try? If so, you're more than welcome to send out your patches and we will review, comments and even merge your contributions.
We are open to the community. And would you please share that what's the most point you want to get from libxcam?

cherixy, 2016-07-22:
@zjamy  I just pay a attention to this project a few days ago becuase of my work. Currently, I have no good  comments. I belive there will be comment with furthur studying this project, and will open to the community.

zjamy, 2016-07-22:
@cherixy no problem.

closed 2017-01-23 by cherixy