intel/libxcam #343: About gamma table setting?

cherixy, 2016-08-04:
According src of libxcam, there are three gamma tables which are  r_gamma，g_gamma and b_gamma. However I can only generate one gamma table with Intel IQ Tool. So, I don't know how I use this table from Intel IQ Tool to set the gamma paramter into libxcam. Are there any points for my problem? Thank you~~

windyuan, 2016-08-04:
most of the time, gamma is set for improve brightness. you can set r/g/b gamma the same table.

cherixy, 2016-08-04:
@windyuan  OK， I get it, Thank you~~  My team wants to use auto-aperture based on libxcam.  I just find that it provides the interface of "set_aperture". This interface can't be used to accomplish auto-aperture. Will libcam support auto-aperture? And can you provide any testing gamma table for me？

windyuan, 2016-08-04:
sorry, set_aperture is not supported by now. actually it depends on driver which is not in our scope.
gamma table you can try gamma=1.5~2.2 and generate a table.

cherixy, 2016-08-04:
@windyuan  Attaced files is generated from IQ Tool based on my imx185_cpf. Where can I modify gamma=1.5~2.2 to generate a table again?
[image]

windyuan, 2016-08-05:
no idea about IQ tool tuning, it's out of our scope. suggest you to contact this tool author or people who give you this tool.
a simple example is that you can write a C program to generate gamma table. you can search online.

cherixy, 2016-08-05:
@windyuan  Thank you very much

closed 2017-01-23 by cherixy

dsrsan, 2017-03-24:
@cherixy  Dear cherixy, please can you provide us the Intel contact who gave you access to the IQ tool?
Is there a way we can get in contact with you via e-mail, we are also working on IMX185.
Best Regards.