intel/libxcam #388: gbe_bin_generater offline compiler occurs  assert error

avansguo, 2016-12-29:
root@kzsmartipc:/tmp# ./gbe_bin_generater kernel_csc_nv12torgba.cl  -o kernel_csc_nv12torgba.cl.bin -t 0x0f31
ASSERTION FAILED: sel.hasDoubleType()
  at file /home/guoqiang/code/git_code/master/IPC/libOpenSource/Trunk/src/Beignet-1.2.1-Source/backend/src/backend/gen_insn_selection.cpp, function void gbe::ConvertInstructionPattern::convertBetweenFloatDouble(gbe::Selection::Opaque&, const gbe::ir::ConvertInstruction&, bool&) const, line 5110
Trace/breakpoint trap

why & thanks

avansguo, 2016-12-29:
./gbe_bin_generater kernel_csc_rgbatonv12.cl -o kernel_csc_rgbatonv12.cl.bin -t 0x0f31   
this is success.   the Beignet version is 1.2.1

zongwave, 2016-12-29:
Could you let us know the beignet version failed to build cl kernel source you used? 

BTW, if you want to compile opencl kernel code into cl binary text, you can try use libxcam tool

test/test-binary-kernel :  source code --> binary
tools/convert-binary-to-text.sh   binary --> binary text

avansguo, 2016-12-29:
the Beignet version is 1.2.1 
libxcam version is 0.7.0

zongwave, 2016-12-29:
with the same Beignet version, "kernel_csc_rgbatonv12.cl" can be built successfully, but "kernel_csc_nv12torgba.cl" built failed? the operation used in these two cl source are almost same, do you make any change in these kernel source?

avansguo, 2016-12-29:
No, I didn't do anything about these kernel source .   
因为我编译好xcam 后， 在我的设备上运行，运行时编译kernel  会报这种错误， 所以才想到先编译试试看， 发现先编译kernel 也会报这种错误。  但是编 kernel_csc_rgbatonv12.cl 就是可以的。

avansguo, 2016-12-29:
i try  "test-binary-kernel ", but it dont work too.  and the errors is same with "gbe_bin_generater"

root@kzsmartipc:/tmp# ./test-binary-kernel -i kernel_csc_nv12torgba.cl -o kernel_csc_nv12torgba.cl.bin
ASSERTION FAILED: sel.hasDoubleType()
  at file /home/guoqiang/code/git_code/master/IPC/libOpenSource/Trunk/src/Beignet-1.2.1-Source/backend/src/backend/gen_insn_selection.cpp, function void gbe::ConvertInstructionPattern::convertBetweenFloatDouble(gbe::Selection::Opaque&, const gbe::ir::ConvertInstruction&, bool&) const, line 5110
Trace/breakpoint trap

guoyejun, 2016-12-30:
How to fix:  
add suffix 'f' for all the literal floating numbers, such as 0.5f, 2.03211f.

Why assertion for kernel_csc_nv12torgba.cl on device 0x0f31:
literal floating numbers without 'f' as suffix in opencl kernel is considered as double (opencl c language is based on c language), while 0x0f31 is the hardware without double support.  
There is no issue if we change to a newer hardware for example 0x1912 with double support.

why kernel_csc_rgbatonv12.cl works on 0x0f31:
it happens to be the case that the kernel is optimized and no double support is needed at hardware level.

avansguo, 2016-12-30:
ok , thank you very much .

avansguo, 2017-01-06:
there is a new error , info is follow:

XCAM WARNING cl_context.cpp:357: CL build program failed on kernel_bayer_pipe, error_code(-11)  build log:stringInput.cl:98:47: error: can't convert between vector values of different size ('float2' (vector of 2 'float' values) and 'double')
stringInput.cl:98:62: error: can't convert between vector values of different size ('float2' (vector of 2 'float' values) and 'double')

guoyejun, 2017-01-06:
could you paste the opencl kernel source?

avansguo, 2017-01-06:
kernel name  : kernel_bayer_pipe

guoyejun, 2017-01-06:
just add two 'f' as below:
 egain[0] = mad(ee, ee_config.ee_gain, out + 0.03f) / (out + 0.03f);

avansguo, 2017-01-06:
i found it too ,  thanks

windyuan, 2017-01-10:
Thank you for @avansguo and @guoyejun pointing out this issue.
We'll check all cl kernels in libxcam to fix the conversion between float and double.

windyuan, 2017-01-10:
Assign this issue to @liuyinhangx .
Thanks,
Wind

liuyinhangx, 2017-01-10:
This issue has been fixed by Zong Wei <wei.zong@intel.com>.
commit bacae1c546a604cd0dcf6cc16c9201083d94e9dd
Thanks.

windyuan, 2017-01-10:
Good. thanks @zongwave.
If verified. please close this issue.

closed 2017-01-10 by avansguo