intel/libxcam #397: fedora 23 libcl not found

arpu, 2017-01-29:
fedora 23 

./configure  --enable-libcl  --enable-opencv --enable-debug 

     libxcam configuration summary
     version                    : 0.8.0
     enable debug               : yes
     enable profiling           : no
     build GStreamer plugin     : no
     build aiq analyzer         : no
     use local aiq              : no
     use local atomisp          : yes
     have opencl lib            : no
     have opencv lib            : yes
     enable 3a lib              : no
     enable smart analysis lib  : no

any help what packages are needed?

uartie, 2017-01-30:
I believe the package is `beignet`.  Try `dnf provides */libcl.so`.

arpu, 2017-01-30:
ok yes you are right but 

export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:$PKG_CONFIG_PATH                                                                                                                                  
echo $PKG_CONFIG_PATH                                                                                                                                                                  
/usr/local/lib/pkgconfig:/usr/lib/pkgconfig
./configure --enable-libcl --enable-opencv
checking for LIBCL... no
checking for LIBCL... no

locate libcl.so                                                                                                                                                                                     
/usr/lib64/beignet/libcl.so

any idea how i can solve this?

uartie, 2017-01-30:
Yup, it looks like neither `beignet` nor `beignet-devel` provides a `libcl.pc` file.  I'm not sure where to get it.  @windyuan can you help with this?

arpu, 2017-02-03:
as workaround 
set HAVE_LIBCL=1 in configure.ac after the libcl check 
and add -lOpenCL to the tests/Makefile

windyuan, 2017-02-07:
@uartie, sorry for late reply, just came back from Chinese New Year vacation. 
yes, you are right, it's the beignet dependency. currently libxcam code depends on <CL/cl_intel.h> which can be get from `beignet-devel`. `libcl.pc` had ever been used in some other projects.  Your code should go to here

    if test "$enable_libcl" = "yes" && test "$HAVE_LIBCL" -eq 0; then
        PKG_CHECK_MODULES(LIBCL, [OpenCL], [HAVE_LIBCL=1], [HAVE_LIBCL=0])
    fi
which means you have to install OpenCL.pc first.

@arpu ,  thanks for your workaround, force to set `HAVE_LIBCL=1` in configure.ac is not necessary if `OpenCL.pc` was installed. could you submit your patch with OpenCL for tests/Makefile?

Thanks,
Wind Yuan

arpu, 2017-02-07:
dnf provides OpenCL.pc                                                                                                                                                                         
Last metadata expiration check: 22:08:54 ago on Mon Feb  6 19:44:30 2017.
Error: No Matches found

about the patch i will look at this!

windyuan, 2017-02-08:
@arpu, thanks.
sorry, I don't have a Fedora OS. In Ubuntu, `OpenCL.pc` is from `ocl-icd-opencl-dev`. you can check ocl-icl related package.

arpu, 2017-04-18:
works now for me i think the problem was not installed ocl dev  package

closed 2017-04-18 by arpu