intel/libxcam #400: XCAM WARNING v4l2_buffer_proxy.cpp:134: unknown v4l2 format(MJPG) to video info

arpu, 2017-02-03:
libxcam cannot capture the MJPG stream 
XCAM INFO v4l2_device.cpp:280: device(/dev/video0) set framerate(48/1)
XCAM INFO v4l2_device.cpp:299: device(/dev/video0) set format(w:2304, h:1536, pixelformat:MJPG, bytesperline:0,image_size:7077888)

it would be perfect if libxcam can use libyami for decode the JPG data with the Intel vaapi Api

windyuan, 2017-02-07:
It's a known issue for a while. There's no bandwidth or resource by now. If I can get your usecase/requirement maybe we can balance priority to support.

arpu, 2017-02-07:
usercase is have 4 usb 3.0 cameras  (360 degree) on one usb 3.1 
bandwidth for raw video is not possible so we use MJPEG compression

windyuan, 2017-02-08:
capture is a one feature in libxcam. as you may know that there are lots features about image processing and video analysis in libxcam. I'm not sure you want use only capture or more other features.
could you send me a mail about your use case and project details? we can give technical support or better advice.

arpu, 2017-02-15:
yes we want use the stitching and image processings from libxam for a live 360 degree streaming video 
what exactly infos do you need?

windyuan, 2017-02-17:
hi, currently stitching based on special parameters and somewhere may have hardcode, e.g. fisheye lens, special input format, resolution and your cameras should aligned or better very close.
capture, you can use gstreamer or other media framework and use libxcam for image processing.
anytime if you have blocks in your trying, you can drop us mail to share your usecase details. we are glad if we can support.

arpu, 2017-03-01:
hi again, i tryed some more gstreamer pipes
gst-launch-1.0 v4l2src device=/dev/video0   !  image/jpeg, width=2304, height=1536, interlaced=false, pixel-aspect-ratio=1/1 ! queue !  vaapidecodebin ! vaapipostproc    !  videoconvert ! xcamfilter copy-mode=1 ! videoconvert ! ximagesink sync=false -v

but i get some errors on this 
most importend part is 
CAM WARNING gstxcamfilter.cpp:706: xcamfilter dequeue buffer failed
0:00:03.453440983 27279      0x122f000 WARN           basetransform gstbasetransform.c:2145:default_generate_output:<xcamfilter0> could not get buffer from pool: ok

the idea is to use vaapi to decode the  image/jpeg and use xcamfilter (opencl)  to  colorspace to 
 NV12 -> to BGR  for opencv usage

arpu, 2017-03-02:
looks like this works 
 gst-launch-1.0 v4l2src io-mode=dmabuf  device=/dev/video0 ! image/jpeg, width=2304, height=1536  ! vaapidecodebin ! xcamfilter copy-mode=1 !  vaapipostproc   ! video/x-raw,format=BGRx  ! ximagesink sync=false -v

windyuan, 2017-03-02:
looks good.
BTY, xcamfilter doesn't export color space conversion up to now. sink and src pad only support NV12 format. vaapidecode link to xcamfilter need a copy since xcamfilter support DMA-fd not VASurfaceID.

closed 2017-04-14 by arpu