intel/libxcam #401: Race condition in Thread::thread_func

Shivantyai, 2017-02-10:
Hello!

I've found a possible race condition in xcam/xcore/xcam_thread.cpp.

```c++
    while (true) {
      {
        SmartLock locker(thread->_mutex);
        if (!thread->_started || ret == false) {
          thread->_started = false;
          thread->_thread_id = 0;
          thread->_exit_cond.signal();
          ret = false;
          break;
        }
      }
      
      ret = thread->loop ();
    }    
    thread->stopped ();
```
When we stop the thread we wait on a condition variable:

```c++
bool Thread::stop ()
{
    SmartLock locker(_mutex);
    if (_started) {
        _started = false;
        _exit_cond.wait(_mutex);
    }
    return true;
}
```
The matter is, that thread->stopped() is not protected with the thread mutex, so there is a possibility for a Thread object to be deleted before thread->stopped() is called, which would lead to a segfault.

The state of the program depends on which thread finishes it's work first( the one deleting the Thread object(and that would lead to a segfault in thread->stopped()) or the one running thread_func() itself).

windyuan, 2017-02-13:
@Shivantyai, Thanks for pointing it out. 
yes, you are right. it may cause crash. we'll fix it ASAP.

windyuan, 2017-02-13:
Hi, I have submitted a patch https://github.com/01org/libxcam/pull/403 to fix this bug.
could you verify whether it is fixed?
Thanks,
Wind

Shivantyai, 2017-02-13:
Hi @windyuan. Yes, I have verified  bug fix - it's ok. Thanks you.

closed 2017-02-13 by Shivantyai