intel/libxcam #411: binary file included with source

jonassmedegaard, 2017-03-10:
The binary file cl_bin/kernel_demo.cl.bin is included with source.

Looks like it was intended to be cleaned in clx_bin/Makefile.am

Some distributors (inlcuding Debian where I am involved) do not allow such binary code as part of source, and your current release 0.8.0 (and HEAD of git as well) therefore in suitable, causing extra work for such (re)distributors in repackaging the source.

Please consider cleaning up the source to not include built binary code.

windyuan, 2017-03-13:
how about keep a clx_bin/kernel_demo.clx.bin file? this can be a text file there and trying to show a example how to write a cl_kernel with binary instead of source-build-in-runtime.

jonassmedegaard, 2017-03-13:
Quoting windyuan (2017-03-13 06:16:18)
> how about keep a clx_bin/kernel_demo.clx.bin file? this can be a text 
> file there and trying to show a example how to write a cl_kernel with 
> binary instead of source-build-in-runtime.

Not sure if that question was for me: I don't understand the code well 
enough to know if possible to replace the binary with a dummy (if that 
was what you meant).

Seems from commit 3af0405 that the binary file was originally compiled 
using a script in tests/ but apparently the build system has since 
evolved to a circular dependency: The binary is required to build the 
libraries, and the libraries are needed to produce the binary :-(

 - Jonas

-- 
 * Jonas Smedegaard - idealist & Internet-arkitekt
 * Tlf.: +45 40843136  Website: http://dr.jones.dk/

 [x] quote me freely  [ ] ask before reusing  [ ] keep private

windyuan, 2017-03-15:
@jonassmedegaard, yes. I'm asking you about the binary in libxcam.
There's a background that, some customer may want to build their own cl kernels and they don't want others to now the algorithms in the kernel. they want to how to support a cl binary without source code. Then libXCam support this feature. As you see, cl_bin/kernel_demo.cl.bin is a example show how to build binary into libXCam.
You report this bug, I think what you mean is that debian distribution doesn't allow binary with source code. and we still want to keep a example how to make cl-binary build-in. Then how about we keep a clx_bin/kernel_demo.clx.bin as text file?

jonassmedegaard, 2017-03-15:
Quoting windyuan (2017-03-15 09:36:28)
> @jonassmedegaard, yes. I'm asking you about the binary in libxcam.
> There's a background that, some customer may want to build their own cl kernels and they don't want others to now the algorithms in [...]

Ah, thanks for spelling it out for me.  Yes, that sounds like a sensible 
solution.

 - Jonas

-- 
 * Jonas Smedegaard - idealist & Internet-arkitekt
 * Tlf.: +45 40843136  Website: http://dr.jones.dk/

 [x] quote me freely  [ ] ask before reusing  [ ] keep private

windyuan, 2017-03-15:
Thank you. we'll fix it soon.

windyuan, 2017-03-23:
fixed by commit 5aad612ed06d2011cdc914555101cf5e01bfd31e
thanks.

liuyinhangx, 2017-03-28:
Verified， close this issue.

closed 2017-03-28 by liuyinhangx