intel/libxcam #424: broken output of test-image-stitching sample

RomanArzumanyan, 2017-04-10:
test-image-stitching sample produces broken output on Ubuntu 16.04 (i7-6770HQ) with Beignet.
Compiled with OpenCL & OpenCV support & "OCL_TILING=0" symbol exported.
Input & output images enclosed.

[image]
[image]

windyuan, 2017-04-11:
@liuyinhangx  will follow up this issue.
what's your cmdline. any more details? we did find this issue in last release.

RomanArzumanyan, 2017-04-12:
./test-image-stitching \
    --input  /home/roman/Videos/input.nv12 \
    --output /home/roman/Videos/output.mp4

Both input & output have default sizes (1920x1080 and 1920x960 corresp.)

liuyinhangx, 2017-04-13:
We can't reproduce this issue, could you let us know the opencv version and  beignet version (or the latest commit id).

RomanArzumanyan, 2017-04-13:
OpenCV 3.2.0
Beignet commit a7dc7692a4f00f33c09a2ad5bc7b58bb9fdd43e5 on master

windyuan, 2017-04-17:
do you have tools to check /home/roman/Videos/input.nv12 is a correct image? I'm thinking about is it possible input.nv12 is already corrupt.

RomanArzumanyan, 2017-04-17:
Hi,
Input isn't broken - actually it's attached to 1st message in thread (converted from nv12 to png).

liuyinhangx, 2017-04-17:
Hi, please try to set OCL_TILING to 0 in the terminal:
$ export OCL_TILING=0

RomanArzumanyan, 2017-04-17:
Hi,
Exporting solves the issue. Thanks a lot.

closed 2017-04-17 by RomanArzumanyan

arpu, 2017-04-18:
hello i get some problems too can you upload the image as nv12? to test for me

windyuan, 2017-04-19:
@arpu if you have png file. you can try gstreamer to convert to NV12.
gst-launch-1.0 filesrc location=in.png ! pngdec  ! videoconvert ! video/x-raw, format=NV12 ! filesink location=out.nv12

if you have Ricoh Theta S camera. you can convert fisheye video mp4 to NV12 by 
gst-launch-1.0 filesrc location=in.mp4 ! qtdemux ! avdec_h264  ! videoconvert ! video/x-raw, format=NV12 ! filesink location=out.nv12

arpu, 2017-04-21:
@windyuan  thx for this hint! works