intel/libxcam #433: Implement iterative blind deblurring algorithm

Andrey1994, 2017-05-14:
This algorithm iterates between updating the restored image and PSF, based on error function minimization

windyuan, 2017-05-15:
looking forward to see this feature. thank you.

windyuan, 2017-06-14:
@Andrey1994 is selected from GSoC (Google Summer of Code https://summerofcode.withgoogle.com) and now is working on this feature. More details from https://github.com/Andrey1994/ImageDeblurring/wiki
Some steps to enable.
1. investigation on deblurring algorithms (done)
2. simulate by matlab or OpenCV (WIP) and tune a good quality on synthetic and real blurring images.
3. porting and clean up code in libxcam with OpenCL.
4. quality and performance tuning.
Thanks,
Wind

Andrey1994, 2017-08-01:
The first version of the IBD algorithm is merged(https://github.com/01org/libxcam/pull/450). Performance and quality improvements required(WIP)

windyuan, 2017-08-02:
Cool. Thanks for @Andrey1994 efforts on IBD enabling.
This feature initiated from GSoC
[libxcam Enable a debluring feature with OpenCL Design](https://summerofcode.withgoogle.com/projects/#4869023098470400)

windyuan, 2017-08-09:
https://github.com/01org/libxcam/pull/454
https://github.com/01org/libxcam/pull/453
merged for debluring framework refine.

excllent123, 2017-09-11:
The result of function edgetaper don't  agree with matlab‘s,  and  I have tested  this function, It doesn't work. Can you update your code? or tell me how it works.  
    Thank  you.

windyuan, 2017-09-11:
@excllent123, could you post the picture from both matlab and libxcam. this can help us to figure out if it's a bug. about the pictures, if not convenient in public, you can also send mail to us (feng.yuan@intel.com and a1994ndrey@gmail.com)
@Andrey1994, could you help check whether it's a bug or something wrong in usage?

Andrey1994, 2017-09-11:
Yep, I will check it. This function just blur the edges with some weights, could you send me how do you use it? 
For now I see one mismatch, I used correlation instead convolution, so for non-symmetric psf it will be differ. I will compare it with matlab results more detailed and fix a code

excllent123, 2017-09-12:
My test code is like this:
    Firstly, I pad the image by using copyMakeBorder in opencv,
    cv::copyMakeBorder(Img,Img,size,size,size,size,cv::BORDER_REPLICATE);
    and then  use  edgetaper
    Img = edgetaper(Img,BlurKernel); 
    and the BlurKernel  is motion-blur-kernel,  
    [0,0,0,0,0
     0 0 0 0 0
     1/5 1/5 1/5 1/5 1/5
     0 0 0 0 0 
     0 0 0 0 0]
   
            I have compared with the output of this function with matlab, but the result is different,  
    and It does'nt work,the result of  image I will send to your email.
           Thank you for your help

Andrey1994, 2017-09-12:
why do you pad image? edgetaper saves the initial sizes. Also Img has to be a single channel image.

I've checked matlab version and libxcam version, there are a little difference in weights calculation, but both version are good for ringing artifacts removing. @windyuan  Have we fix weights initialization?

windyuan, 2017-09-12:
@Andrey1994 there's only one rule for weights(matlab version). we need fix. Else, keep the current version.

Andrey1994, 2017-09-12:
@excllent123 Matlab calculates the autocorrelation of PSF in another and in my opinion rather strange way. They use 1d autocorrelation of vector which is a sum of each PSF rows\columns. In libxcam we use 2d autocorrealtion of PSF. That's why their weights are differ

Maybe their implementation outperform our, but I think both of them can be used. So, I think at least for a while we will keep current version.

excllent123, 2017-09-13:
@Andrey1994   Thanks  for your help.

windyuan, 2018-04-13:
close this feature since already released in 1.1.0

closed 2018-04-13 by windyuan