intel/libxcam #449: Fail in test-cl-image

lenovoivc, 2017-07-19:
Try to compile beignet and libxcam on Ubuntu 16.04 / CUDA 8.0 
Finally, errors return, when trying:

# ./test-cl-image -t retinex -f RGBA -i ./seamless-blend.png -o ./semless.png
/libxcam/tests/.libs/lt-test-cl-image: /usr/local/cuda-8.0/lib64/libOpenCL.so.1: no version information available (required by /libxcam/modules/ocl/.libs/libxcam_ocl.so.1)
/libxcam/tests/.libs/lt-test-cl-image: /usr/local/cuda-8.0/lib64/libOpenCL.so.1: no version information available (required by /libxcam/modules/ocl/.libs/libxcam_ocl.so.1)
/libxcam/tests/.libs/lt-test-cl-image: /usr/local/cuda-8.0/lib64/libOpenCL.so.1: no version information available (required by /libxcam/modules/ocl/.libs/libxcam_ocl.so.1)
/libxcam/tests/.libs/lt-test-cl-image: /usr/local/cuda-8.0/lib64/libOpenCL.so.1: no version information available (required by /libxcam/modules/ocl/.libs/libxcam_ocl.so.1)
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
cl_get_gt_device(): error, unknown device: 0
XCAM WARNING cl_device.cpp:113: get cl device ID failed
XCAM WARNING cl_device.cpp:52: CL device init failed
XCAM DEBUG cl_device.cpp:54: CL device constructed
lt-test-cl-image: cl_kernel.cpp:39: XCam::CLKernel::CLKernel(const XCam::SmartPtr<XCam::CLContext>&, const char*): Assertion `context.ptr ()' failed.
Aborted (core dumped)
------------------------------------------------------------------------------------

Actually, errors below have occurred during Make:

make  all-recursive
make[1]: Entering directory '/libxcam'
Making all in xcore
make[2]: Entering directory '/libxcam/xcore'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/libxcam/xcore'
Making all in clx_kernel
make[2]: Entering directory '/libxcam/clx_kernel'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/libxcam/clx_kernel'
Making all in modules
make[2]: Entering directory '/libxcam/modules'
Making all in ocl
make[3]: Entering directory '/libxcam/modules/ocl'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/libxcam/modules/ocl'
make[3]: Entering directory '/libxcam/modules'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/libxcam/modules'
make[2]: Leaving directory '/libxcam/modules'
Making all in plugins
make[2]: Entering directory '/libxcam/plugins'
make[3]: Entering directory '/libxcam/plugins'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/libxcam/plugins'
make[2]: Leaving directory '/libxcam/plugins'
Making all in wrapper
make[2]: Entering directory '/libxcam/wrapper'
make[3]: Entering directory '/libxcam/wrapper'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/libxcam/wrapper'
make[2]: Leaving directory '/libxcam/wrapper'
Making all in capi
make[2]: Entering directory '/libxcam/capi'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/libxcam/capi'
Making all in tests
make[2]: Entering directory '/libxcam/tests'
/bin/bash ../libtool  --tag=CXX   --mode=link g++ -fPIC -DSTDC99 -W -Wall -D_REENTRANT -Wformat -Wformat-security -fstack-protector -g -DDEBUG -std=c++0x -I/usr/include/libdrm -ldrm -I../ext/atomisp -I../xcore -I../modules  -g -O2   -o test-device-manager test_device_manager-test-device-manager.o ../xcore/libxcam_core.la  ../modules/ocl/libxcam_ocl.la 
libtool: link: g++ -fPIC -DSTDC99 -W -Wall -D_REENTRANT -Wformat -Wformat-security -fstack-protector -g -DDEBUG -std=c++0x -I/usr/include/libdrm -I../ext/atomisp -I../xcore -I../modules -g -O2 -o .libs/test-device-manager test_device_manager-test-device-manager.o  -ldrm ../xcore/.libs/libxcam_core.so ../modules/ocl/.libs/libxcam_ocl.so -pthread -Wl,-rpath -Wl,/usr/local/lib/beignet/lib
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clCreateCommandQueueWithProperties@OPENCL_2.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetDeviceIDs@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clReleaseCommandQueue@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clBuildProgram@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetExtensionFunctionAddressForPlatform@OPENCL_1.2'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clCreateKernel@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetPlatformIDs@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clSetKernelArg@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clCreateProgramWithBinary@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetPlatformInfo@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetProgramInfo@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clCreateBuffer@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clReleaseProgram@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clSetEventCallback@OPENCL_1.1'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clReleaseEvent@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clEnqueueReadBuffer@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clCreateContext@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clCreateImage@OPENCL_1.2'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clWaitForEvents@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clEnqueueMapImage@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clEnqueueWriteBuffer@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clFinish@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetProgramBuildInfo@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetEventInfo@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clReleaseContext@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clEnqueueUnmapMemObject@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetDeviceInfo@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clEnqueueNDRangeKernel@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clCreateProgramWithSource@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetImageInfo@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clReleaseKernel@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clReleaseMemObject@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clFlush@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clGetMemObjectInfo@OPENCL_1.0'
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clEnqueueMapBuffer@OPENCL_1.0'
collect2: error: ld returned 1 exit status
Makefile:541: recipe for target 'test-device-manager' failed
make[2]: *** [test-device-manager] Error 1
make[2]: Leaving directory '/libxcam/tests'
Makefile:436: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/libxcam'
Makefile:368: recipe for target 'all' failed
make: *** [all] Error 2

windyuan, 2017-07-19:
I see  /usr/local/cuda-8.0/lib64/libOpenCL.so.1. currently libxcam doens't support nvidia, though we'are trying to make libxcam common without HW dependencies.  
Make sure ocl-icd-dev, ocl-icd-opencl-dev installed.
also check whether beignet path is in /etc/OpenCL/vendors

lenovoivc, 2017-07-19:
Thank you, Yuan!
Confirm that:
1) ocl-icd-dev and ocl-icd-opencl-dev have already been installed.
2) In /etc/OpenCL/vendors, there has been a file named intel-beignet.icd, which contains a single line of "/usr/local/lib/beignet//libcl.so"

Moreover, libOpenCL.so locates both at
/usr/local/cuda-8.0/lib64
and
/usr/lib/x86_64-linux-gnu

How can I get rid of dependency on CUDA?

windyuan, 2017-07-19:
I think cuda also has a ICD file in /etc/OpenCL/vendors. you can 
 $ export OCL_ICD_VENDORS=/etc/OpenCL/vendors/intel-beignet.icd 
or move cuda icd to somewhere(backup the file first) else, then try again.

lenovoivc, 2017-07-19:
Successfully made libxcam after removing CUDA path from LD_LIBRARY_PATH
However, the erros below still exist when calling test-cl-image 
X server found. dri2 connection failed!
 DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
 Assuming 131072kB available aperture size.
 May lead to reduced performance or incorrect rendering.
 get chip id failed: -1 [22]
 param: 4, val: 0
 DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
 Assuming 131072kB available aperture size.
 May lead to reduced performance or incorrect rendering.
 get chip id failed: -1 [22]
 param: 4, val: 0
 X server found. dri2 connection failed!
 DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
 Assuming 131072kB available aperture size.
 May lead to reduced performance or incorrect rendering.
 get chip id failed: -1 [22]
 param: 4, val: 0
 DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
 Assuming 131072kB available aperture size.
 May lead to reduced performance or incorrect rendering.
 get chip id failed: -1 [22]
 param: 4, val: 0
 X server found. dri2 connection failed!
 DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
 Assuming 131072kB available aperture size.
 May lead to reduced performance or incorrect rendering.
 get chip id failed: -1 [22]
 param: 4, val: 0
 DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
 Assuming 131072kB available aperture size.
 May lead to reduced performance or incorrect rendering.
 get chip id failed: -1 [22]
 param: 4, val: 0
 cl_get_gt_device(): error, unknown device: 0
 XCAM WARNING cl_device.cpp:113: get cl device ID failed
 XCAM WARNING cl_device.cpp:52: CL device init failed
 XCAM DEBUG cl_device.cpp:54: CL device constructed
 lt-test-cl-image: cl_kernel.cpp:39: XCam::CLKernel::CLKernel(const XCam::SmartPtrXCam::CLContext&, const char*): Assertion `context.ptr ()' failed.
 Aborted (core dumped)

windyuan, 2017-07-20:
you may need to change to 'root' to test or stop service gdm/lightdm. libxcam use libdrm for buffer-management and it's possible the request of buffer-manager failed in authentication with XServer.
` X server found. dri2 connection failed!`

lenovoivc, 2017-07-20:
Shut down lightdm in command line mode by 
sudo servie ligthtdm stop
and then called
sudo ./test-cl-image -t retinex -f RGBA -i ./input.png -o ./output.png
got error below:

DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)

windyuan, 2017-07-20:
It's more like Beignet doens't support your HW. you can get Beignet code, and there's a script 'GetGenID.sh' which you can run to see your platform.
Beignet supports (cat GetGenID.sh)

        #IVB
        genpciid=(0152 0162 0156 0166 015a 016a)
        #BYT
        genpciid+=(0f31)
        #HSW
        genpciid+=(0402 0412 0422 040a 041a 042a 0406 0416 0426 040b 041b 042b 040e 041e 042e)
        genpciid+=(0c02 0c12 0c22 0c0a 0c1a 0c2a 0c06 0c16 0c26 0c0b 0c1b 0c2b 0c0e 0c1e 0c2e)
        genpciid+=(0a02 0a12 0a22 0a0a 0a1a 0a2a 0a06 0a16 0a26 0a0b 0a1b 0a2b 0a0e 0a1e 0a2e)
        genpciid+=(0d02 0d12 0d22 0d0a 0d1a 0d2a 0d06 0d16 0d26 0d0b 0d1b 0d2b 0d0e 0d1e 0d2e)
        #BRW
        genpciid+=(1602 1606 160a 160d 160e 1612 1616 161a 161d 161e 1622 1626 162a 162d 162e)
        #BSW
        genpciid+=(22b0 22b1 22b2 22b3)
        #SKL
        genpciid+=(1906 1916 1926 190e 191e 1902 1912 1932 190b 191b 192b 193b 190a 191a 192a 193a)
        #BXT
        genpciid+=(5a84)
        pciid=($(lspci -nn | grep "\[8086:.*\]" -o | awk -F : '{print $2}' | awk -F ] '{print $1}'))

windyuan, 2017-07-20:
BTW, ./test-cl-image -t retinex -f RGBA -i ./input.png -o ./output.png. 
currently most features' supported format is NV12. you can use ffmpeg or gstreamer change to NV12 format for input.  and ouput is same.  if you image is not 1080P, you need add -W -H to specify width/height. e.g.
./test-cl-image -t retinex -f NV12 -i ./input.nv12 -o ./output.nv12 -W 1920 -H 1080

lenovoivc, 2017-07-20:
What a platform is required?
Got nothing when running GetGenID.sh in my PC.
I changed format and ran
./test-cl-image -t retinex -f NV12 -i ./input.nv -o ./output.nv -W 512 -H 512
but still the same error:
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)

windyuan, 2017-07-20:
what's your platform? from Beignet GetGenID.sh. we can see it supports IVB, BYT, HSW, BRW, BSW, SKL, BXT(APL)

lenovoivc, 2017-07-20:
My two PCs with pciid of 191f and 0e00 both seem not to be supported within GetGenID.
May I manually generate ID for my PCs? Or, I will have to try more PCs.

windyuan, 2017-07-21:
I'm not sure your platform from codeid. If you are sure you have another PC with any of IVB, BYT, HSW, BRW, BSW, SKL, BXT(APL). please try again.
I also suggest you to drop a mail to beignet mailing list(beignet@lists.freedesktop.org), maybe your pc can be supported but they didn't add the code into beignet.

lenovoivc, 2017-07-21:
Tried more than 5 PCs, but they are all not supported. And sent my problem to beignet@lists.freedesktop.org
If possible, a list of supported PC modes could be given at the webpage. IVB, BYT, HSW ... look too professional to me, a non-Intel person.

closed 2017-07-21 by lenovoivc

windyuan, 2017-07-21:
@lenovoivc  sorry for inconvenience, we are planning to add more drivers support in next release.

lenovoivc, 2017-07-21:
Thank you. Hope to test libxcam in my PCs soon.
BTW: pciid seems related to I/O rather than chip. If testing on local images instead of camera, do I/O drivers matter?

windyuan, 2017-07-21:
here is what lspci -nn showed on my PC with Intel integrated graphics controller. Beignet is an OCL driver based on Intel integrated graphics card (chip).
`00:02.0 VGA compatible controller [0300]: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller [8086:0416] (rev 06)`
libxcam is independent from camera. and processed by GPU via OpenCL. The result is same either on camera or on local image.