intel/libxcam #456: compile failure on ubuntu trusty 14.04 LTS

uartie, 2017-08-15:
On Ubuntu Trusty 14.04 LTS, the following compilation failure occurs:

```
make[2]: Entering directory `/opt/media/build/libxcam/tests'
/bin/bash ../libtool  --tag=CXX   --mode=link g++ -fPIC -DSTDC99 -W -Wall -D_REENTRANT -Wformat -Wformat-security -fstack-protector -g -DDEBUG -std=c++0x -I/usr/include/libdrm   -ldrm   -I../ext/atomisp -I../xcore -I../modules  -g3  -L/opt/media/install/lib -o test-device-manager test_device_manager-test-device-manager.o ../xcore/libxcam_core.la  ../modules/ocl/libxcam_ocl.la 
libtool: link: g++ -fPIC -DSTDC99 -W -Wall -D_REENTRANT -Wformat -Wformat-security -fstack-protector -g -DDEBUG -std=c++0x -I/usr/include/libdrm -I../ext/atomisp -I../xcore -I../modules -g3 -o .libs/test-device-manager test_device_manager-test-device-manager.o  -ldrm -L/opt/media/install/lib ../xcore/.libs/libxcam_core.so ../modules/ocl/.libs/libxcam_ocl.so -pthread -Wl,-rpath -Wl,/opt/media/install/lib
../modules/ocl/.libs/libxcam_ocl.so: undefined reference to `clCreateCommandQueueWithProperties'
collect2: error: ld returned 1 exit status
make[2]: *** [test-device-manager] Error 1
make[2]: Leaving directory `/opt/media/build/libxcam/tests'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/opt/media/build/libxcam'
make: *** [all] Error 2
```

I have installed apt packages **ocl-icd-dev**, **ocl-icd-opencl-dev** and compiled/installed beignet from source.

liuyinhangx, 2017-08-16:
Could you let us know the beignet version (or the latest commit id)?

uartie, 2017-08-16:
beignet (master) heads/master-0-g757ccf02d6ea

uartie, 2017-08-16:
The problem is that on ubuntu trusty, `apt install ocl-icd-dev ocl-icd-opencl-dev` produces:

```
Reading package lists...
Building dependency tree...
Reading state information...
The following extra packages will be installed:
  ocl-icd-libopencl1 opencl-headers
Suggested packages:
  opencl-icd
The following NEW packages will be installed:
  ocl-icd-dev ocl-icd-libopencl1 ocl-icd-opencl-dev opencl-headers
0 upgraded, 4 newly installed, 0 to remove and 2 not upgraded.
Need to get 102 kB of archives.
After this operation, 739 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu/ trusty/universe ocl-icd-libopencl1 amd64 2.1.3-4 [28.1 kB]
Get:2 http://archive.ubuntu.com/ubuntu/ trusty/universe ocl-icd-dev amd64 2.1.3-4 [9842 B]
Get:3 http://archive.ubuntu.com/ubuntu/ trusty/universe opencl-headers all 1.2-2013.10.23-1 [61.2 kB]
Get:4 http://archive.ubuntu.com/ubuntu/ trusty/universe ocl-icd-opencl-dev amd64 2.1.3-4 [2430 B]
Fetched 102 kB in 0s (185 kB/s)
Selecting previously unselected package ocl-icd-libopencl1:amd64.
(Reading database ... 30818 files and directories currently installed.)
Preparing to unpack .../ocl-icd-libopencl1_2.1.3-4_amd64.deb ...
Unpacking ocl-icd-libopencl1:amd64 (2.1.3-4) ...
Selecting previously unselected package ocl-icd-dev.
Preparing to unpack .../ocl-icd-dev_2.1.3-4_amd64.deb ...
Unpacking ocl-icd-dev (2.1.3-4) ...
Selecting previously unselected package opencl-headers.
Preparing to unpack .../opencl-headers_1.2-2013.10.23-1_all.deb ...
Unpacking opencl-headers (1.2-2013.10.23-1) ...
Selecting previously unselected package ocl-icd-opencl-dev:amd64.
Preparing to unpack .../ocl-icd-opencl-dev_2.1.3-4_amd64.deb ...
Unpacking ocl-icd-opencl-dev:amd64 (2.1.3-4) ...
Processing triggers for man-db (2.6.7.1-1ubuntu1) ...
Setting up ocl-icd-libopencl1:amd64 (2.1.3-4) ...
Setting up ocl-icd-dev (2.1.3-4) ...
Setting up opencl-headers (1.2-2013.10.23-1) ...
Setting up ocl-icd-opencl-dev:amd64 (2.1.3-4) ...
Processing triggers for libc-bin (2.19-0ubuntu6.13) ...
```

... beignet installation from source overwrites the apt `opencl-headers` package files.  The beignet header files define CL_VERSION_2_0.  However, the libOpenCL.so from apt package `ocl-icd-libopencl1` was likely compiled with the `opencl-headers (1.2-2013.10.23-1)` apt package which is not version 2.0 and therefore does not have `clCreateCommandQueueWithProperties` defined in the libOpenCL.so.

First of all, beignet is breaking rules by installing its own opencl headers.  Horrible practice!  Next, why is opencl-headers package on trusty at version 1.2 whereas other related packages are at 2.1.3?  Is that a trusty package maintainer mishap?  Nonetheless, following the libxcam and beignet build instructions does not work for trusty.

windyuan, 2017-08-16:
@uartie, I ever get similar issue. It's caused by beignet installed ocl headers which conflict with ocl-icd headers.
There's a workaround, you can make-install beignet to other directory like  --prefix=/opt/beignet/ instead of default --prefix=/usr/local

we add patch to support dependency of cl_intel.h
* 4bd7e36 - build: download cl_intel.h for local comipilation (5 days ago) <Yinhang Liu>

uartie, 2017-08-16:
Why would you force users to do this?  That is a lame hack and does nothing to solve the real issue and just introduces bugs!

windyuan, 2017-08-17:
yes, you are right. but I think a perfect fix should be in beignet. It installed different version of OCL headers which may conflict ocl-icd headers. I don't think libxcam is the only project got this issue. we can report a bug there.
Meanwhile, libxcam is trying to remove all dependencies with specific ocl drivers.

uartie, 2017-08-22:
Can you please work with beignet maintainers to resolve this problem?

windyuan, 2017-08-23:
I have reported bug for beignet https://bugs.freedesktop.org/show_bug.cgi?id=102362 .

windyuan, 2017-08-23:
Beignet maintainer suggest to manually upgrade ocl-icd-libopencl and opencl-headers in 14.04. make sure opencl API versions >2.0.
1. upgradte ocl-icd-libopencl1 > 2.0. yours is already 2.1.x
2. opencl-headers. version 2.0. yours is 1.2. need upgrade.

uartie, 2017-08-23:
The distro does not provide 2.1.x opencl-headers.

uartie, 2017-08-23:
Also, note that ocl-icd-libopencl1 (package version 2.1.3) actually only provides libopencl-1.2:
```
Package: ocl-icd-libopencl1
Status: install ok installed
Priority: extra
Section: libs
Installed-Size: 131
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Multi-Arch: same
Source: ocl-icd
Version: 2.1.3-4
Replaces: amd-app, libopencl1, nvidia-libopencl1-dev
Provides: libopencl-1.1-1, libopencl-1.2-1, libopencl1
Depends: libc6 (>= 2.14)
Pre-Depends: multiarch-support
Suggests: opencl-icd
Conflicts: amd-app, libopencl1, nvidia-libopencl1-dev
Description: Generic OpenCL ICD Loader
 OpenCL (Open Computing Language) is a multivendor open standard for
 general-purpose parallel programming of heterogeneous systems that include
 CPUs, GPUs and other processors.
 .
 This package contains an installable client driver loader (ICD Loader)
 library that can be used to load any (free or non-free) installable client
 driver (ICD) for OpenCL. It acts as a demultiplexer so several ICD can
 be installed and used together.
Original-Maintainer: Vincent Danjean <vdanjean@debian.org>
Homepage: https://forge.imag.fr/projects/ocl-icd/
```

uartie, 2017-08-23:
So I installed ocl-icd from source into `--prefix=/usr` and turns out there are still issues with linking.  Ubuntu Trusty's version of libtool has `link_all_deplibs=no` by default (this has been a common problem and debate for many developers).  Thus, implicit dependencies are not added to the linker command.  As you can see in the output listed in the initial description, the linker is missing `-lOpenCL` and the library path to find it.  This is a bug in the tests/Makefile.am.  For this scenario all libraries (direct or indirect to .la libs) should be listed in Makefile.am.

On other distros, libtool has `link_all_deplibs=unknown/yes` by default and therefore when linking to libxcam_ocl.la, the libraries it linked are implicitly included in the linker command.

uartie, 2017-08-23:
Please disregard last comment... I forgot to remove distro ocl-icd packages.  Installing from source works fine.

uartie, 2017-08-23:
Installing ocl-icd from source seems to work fine.  Thus, closing this issue for now.

Beignet still needs deal with the problem from their side, however.  Thanks for filing a bug with them.

closed 2017-08-23 by uartie

windyuan, 2017-08-24:
Good to see your env start working.

mualloc, 2017-11-08:
I could not really understand and apply succesfully what you are describing here so in /libxcam/modules/ocl/cl_context.cpp I added
<#undef CL_VERSION_2_0>
before
<#if defined (CL_VERSION_2_0) && (CL_VERSION_2_0 == 1)
    cmd_queue_id = clCreateCommandQueueWithProperties (_context_id, device_id, 0, &err_code);>

and it has been successfully built and installed, though I have not write and run any program, yet.
So, what do you think, can my change be regarded as legal?

windyuan, 2017-11-08:
@mualloc, it's a workaround.  it's better you can upgrade ocl-icd and header files to 2.0 version. although we use very few 2.0 features like clCreateImage2D from clBuffer or clSubBuffer, if your driver doesn't support these features, your workaround can not work and program failed in runtime.

mualloc, 2017-11-08:
@windyuan How can I upgrade it? I am novice to both Linux and OpenCL.

windyuan, 2017-11-08:
you can get source code from
from https://github.com/OCL-dev/ocl-icd
then build and install to replace current version.
If your Ubuntu version >=16.04, you can directly upgrade by apt-get