intel/libxcam #473: Don't use "using namespace ..." in header files

uartie, 2017-09-11:
"using namespace" in header files is bad.  There are lots of resources on the internet that explain why it is bad.  For example, https://stackoverflow.com/questions/4872373/why-is-including-using-namespace-into-a-header-file-a-bad-idea-in-c

Remove all instances of "using namespace" from header files:

```
for i in `find . | grep "\.h"`; do grep -n "using namespace" $i; if [ $? -eq 0 ]; then echo "^"$i"^"; echo; fi; done
```

windyuan, 2017-09-12:
Thanks for the reminder.
We take care of the exported header files without 'using namespace'.

from the script search. I got results

		29:using namespace XCam;
		^./wrapper/gstreamer/gstxcambuffermeta.h^
		
		29:using namespace XCam;
		30:using namespace GstXCam;
		^./wrapper/gstreamer/gstxcambufferpool.h^
		
		26:using namespace XCam;
		^./wrapper/gstreamer/gst_xcam_utils.h^
		
		28:using namespace XCam;
		29:using namespace GstXCam;
		^./wrapper/gstreamer/gstxcamsrc.h^
		
		30:using namespace XCam;
		31:using namespace GstXCam;
		^./wrapper/gstreamer/gstxcamfilter.h^
		
		30:using namespace cv;
		31:using namespace cv::videostab;
		32:using namespace std;
		^./plugins/smart/dvs/libdvs/stabilizer.h^
		
		27:using namespace XCam;
		^./tests/test_inline.h^
		
		31:using namespace XCam;
		^./capi/context_priv.h^

all gstxcam*.h are for gst-plugins, headers are for internal usage.
stabilizer.h is a plugin.
test_inline.h is for all tests.
context_priv.h is private header, actually like .cpp files.

we can remove all 'using namespace' in gstxcam*.h and stabilizer.h.
But want to keep test_inline.h and context_priv.h. is CI build allowed?

windyuan, 2017-09-12:
@zongwave , please fix this issue.

windyuan, 2017-10-27:
it's fixed. close this issue.
any question, we can reopen.

closed 2017-10-27 by windyuan