intel/libxcam #474: Configure should check minimum OpenCV version

uartie, 2017-09-11:
Configure allows any opencv version.  However, libxcam will not compile with opencv < 3.0 when opencl enabled:

```
In file included from ../../modules/ocl/cv_base_class.h:32:0,
                 from ../../modules/ocl/cv_feature_match.h:31,
                 from cl_image_360_stitch.h:30,
                 from cl_post_image_processor.cpp:35:
../../modules/ocl/cv_context.h:39:32: fatal error: opencv2/core/ocl.hpp: No such file or directory
 #include <opencv2/core/ocl.hpp>
```

`opencv2/core/ocl.hpp` is not available until opencv 3.0.

Configure should require opencv at least version 3.0

windyuan, 2017-09-12:
assign to @zongwave

windyuan, 2017-09-14:
fixed by https://github.com/01org/libxcam/pull/479

uartie, 2017-09-14:
If user explicitly passes `--enable-opencv` to configure, then configure should fail if opencv < 3.0.

#479 disables opencv unconditionally if < 3.0 irregardless of `--enable-opencv` flag explicitly specified.

You can reference libva's configure.ac for examples on how to deal with this kind of case in configure.ac.

windyuan, 2017-09-15:
@uartie you are right. sorry for the quick patch merge.
@zongwave it's not necessary to define lots small opencv versions in configure. it's just a dependence lib.  and we can't only support opencv 3.0. Most of our features can come from 2.4.
       - m4_define([opencv_major_version_req], [3])
       - m4_define([opencv_minor_version_req], [0])
       - m4_define([opencv_revision_version_req], [0])

uartie, 2017-09-15:
@windyuan IIUC Open**CV** >= 3.0 is only needed if enabling Open**CL**?  Otherwise, Open**CV** >= 2.4 is sufficient?

windyuan, 2017-09-18:
@uartie  Yes, you can think so. It's a little complicated we will support only CPU version if OpenCV <3.0

windyuan, 2017-10-27:
minimum version still keeps 3.0.
but there's CAPI version supported for obsolete OpenCV.  fixed in https://github.com/01org/libxcam/pull/495

windyuan, 2018-01-30:
Bug has been fixed for a while. close this bug.
libxcam Linux build will not support <3.0 opencv since some important feature missing and API incompatible.
libxcam Android build can support the native opencv (platform/external/opencv) which has a version number (1.0)

closed 2018-01-30 by windyuan