intel/libxcam #508: capi Makefile.am missing OpenCV CFLAGS

ngaloppo, 2017-11-06:
Needs a `#(OPENCV_CFLAGS)` line here: https://github.com/01org/libxcam/blob/master/capi/Makefile.am#L9

windyuan, 2017-11-07:
thanks for pointing this out. @liuyinhangx please fix.

liuyinhangx, 2017-11-07:
@ngaloppo 
Could you please provide the error log?
I can't reproduce this issue on my platform(Ubuntu 16.04), it seems that capi module does not use opencv, thanks.

ngaloppo, 2017-11-07:
See below. Looks like `cl_image_360_stitch.h` uses OpenCV. 

```
In file included from ../modules/ocl/cv_base_class.h:32:0,
                 from ../modules/ocl/cv_feature_match.h:31,
                 from ../modules/ocl/cl_image_360_stitch.h:30,
                 from context_priv.cpp:32:
../modules/ocl/cv_context.h:38:30: fatal error: opencv2/opencv.hpp: No such file or directory
compilation terminated.
Makefile:516: recipe for target 'libxcam_capi_la-context_priv.lo' failed
```

closed 2018-12-05 by zongwave