intel/libxcam #517: ‘XCAM_VERSION’ was not declared in this scope

mualloc, 2017-11-09:
Hello,
I am trying to compile "test-image-deblurring.cpp" in tests folder of libxcam. However, compiler asserts these error lines:

In file included from /usr/include/xcam/ocl/cl_image_handler.h:33:0,
                 from /usr/include/xcam/ocl/cl_blender.h:27,
                 from test-image-deblurring.cpp:28:
/usr/local/include/xcam/x3a_result.h: In constructor ‘XCam::X3aStandardResultT<StandardResult>::X3aStandardResultT(uint32_t, XCamImageProcessType, uint32_t)’:
/usr/local/include/xcam/x3a_result.h:117:33: error: ‘XCAM_VERSION’ was not declared in this scope
         _result->head.version = XCAM_VERSION;
                                 ^
I searched XCAM_VERSION by typing grep -nr 'XCAM_VERSION' under libxcam folder and the result is 

x3a_result.h:117:        _result->head.version = XCAM_VERSION;

So, I think this macro is never defined. Is this so and how can I solve this problem?

liuyinhangx, 2017-11-10:
@mualloc 
Please check that the automake and autoconf have been installed correctly, and try to rerun autogen.sh.

windyuan, 2018-01-30:
This issue was already fixed. 
I need close this bug and if anyone found XCAM_VERSION issues in Android or Linux, please reopen.
Thanks.

closed 2018-01-30 by windyuan