intel/libxcam #569: libxcam build failure with ToT opencv

jayesht, 2018-04-12:
The build is failing with ToT https://github.com/opencv/opencv/commits/master branch on Fedora and Ubuntu. 
Error snippet:
14:11:20,466 INFO  - In file included from soft_stitcher.cpp:34:0:
14:11:20,466 INFO  - cv_capi_feature_match.h:59:29: error: 'CvArr' has not been declared
14:11:20,466 INFO  - void add_detected_data (CvArr* image, std::vector<CvPoint2D32f> &corners);
14:11:20,466 INFO  - ^~~~~
14:11:20,466 INFO  - cv_capi_feature_match.h:59:55: error: 'CvPoint2D32f' was not declared in this scope
14:11:20,466 INFO  - void add_detected_data (CvArr* image, std::vector<CvPoint2D32f> 

Full log attached.
[libxcam_fail.txt](https://github.com/intel/libxcam/files/1903799/libxcam_fail.txt)

Error is caused by OpenCV dropping support for DISABLE_OPENCV_24_COMPATIBILITY
https://github.com/opencv/opencv/commit/2b2fa58f977b3f5334d18cba381e568e7e5622ed

windyuan, 2018-04-13:
I'll check this issue. thanks for your report.

windyuan, 2018-04-13:
Fixed. please try build again.

jayesht, 2018-04-13:
Worked with opencv ToT. Thanks.

closed 2018-04-13 by jayesht