intel/libxcam #571: Are there any test files for testing?

lyyiangang, 2018-04-25:
Thanks for your repo, It's powerful and amazing. I am try it's stitch function. But I find there are no test video (.nv12) can be used.  And I am not sure whether a calibration board should be captured in the test video. 

Thanks
Sir.

Po-Jen, 2018-04-26:
Similar question here! Do you have sample video (input.nv12) and gyro data (gyro-data.csv) for gyroscope 3-DoF (orientation) based video stabilization?

Thanks in advance!

windyuan, 2018-04-26:
@Po-Jen,
yes. gyro data and video stabilization clips can be shared to you. we don't have public place to store the big data. please send a mail to @zongwave  and CC me. @zongwave  can share his sample data to you.

Thanks,
Wind Yuan

zongwave, 2018-04-26:
@Po-Jen you can get video & gyro data from https://github.com/zongwave/IPASS/tree/master/VideoStab/data

please convert mp4 file to nv12 file by yourself, you can use ffmpeg to do conversion
ffmepg -i input.mp4 -an -pix_fmt nv12 out.yuv

Po-Jen, 2018-04-27:
@windyuan @zongwave Thank you so much!!

closed 2018-12-05 by zongwave