intel/libxcam #588: libxcam compile fails since opencv commit 9097ccf515b8

uartie, 2018-09-13:
libxcam fails to compile with recent upstream opencv commit:
```
commit 9097ccf515b8a92ad39808d01d9ee55f5a4985d3
Author: Alexander Alekhin <alexander.a.alekhin@gmail.com>
Date:   Sun Sep 9 21:46:46 2018 +0000

    drop legacy headers
```
compile output:
```
<snip>
In file included from cv_capi_feature_match.cpp:23:0:
cv_capi_feature_match.h:33:10: fatal error: cv.hpp: No such file or directory
 #include <cv.hpp>
          ^~~~~~~~
compilation terminated.
make[3]: *** [Makefile:610: libxcam_soft_la-cv_capi_feature_match.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from soft_stitcher.cpp:34:0:
cv_capi_feature_match.h:33:10: fatal error: cv.hpp: No such file or directory
 #include <cv.hpp>
          ^~~~~~~~
compilation terminated.
make[3]: *** [Makefile:603: libxcam_soft_la-soft_stitcher.lo] Error 1
<snip>
```

windyuan, 2018-09-14:
Thanks @uartie for reporting this issue.
assign to @liuyinhangx . 
thanks.

windyuan, 2018-09-20:
@uartie ,
Would you mind change opencv to 3.4.3 release tag or 3.4 branch?
opencv APIs changes so fast recently. The version is 4.0+.  opencv maintainer probably plan a big framework change and all the header files would be changed from dir <opencv> to <opencv4> from latest commit.  There would be a big gap between 3.4 and 4.0 in build.
We donot want to take too much risk to guess 4.0 changes before its official release. Then it's better to fallback and stay on 3.4.3 version at this moment.

This problem blocked https://github.com/intel/libxcam/pull/587. I appreciate you to take time supporting us fix CI build.

Thanks,
Wind

uartie, 2018-09-20:
@windyuan I have temporarily changed CI to track opencv 3.4 branch.  If you don't plan to support opencv 4.x+ anytime in the near future, then I recommend you update readme and wiki about opencv 3.4 requirement.  Also, you may add a check during autoconf to fail if opencv >= 4.x for people that rarely read the documentation.

windyuan, 2018-09-25:
@uartie,
Thanks for your support and suggestions. We'll evaluate opencv 4.0 changes after official release, then decide to move on or stay. @liuyinhangx will change the patch to 3.4 as the maximum version in configure at this moment.  Thanks for your suggestion. Wiki page about opencv version recommendation, we also have concerns about performance regression between 3.1 and 3.4. We'll update it later.

Thanks,
Wind

closed 2018-11-23 by zongwave