intel/libxcam #616: what is the unit of extrinsic parameters?

simpzan, 2019-03-27:
Hi, there. 
I am trying to run test-surround-view with following extrinsic parameters, it doesn't work. 
the parameters are calculated with opencv `solvePnP()`. what is the expected unit/format for the extrinsic parameters? and how to obtain them? Thanks!
```
#Position x
-0.03677052527289915;
#Position y
0.8791858435421135;
#Position z
0.08443662467164369
#Rotation x (roll)
-0.9773921430544583
#Rotation x (pitch)
0.02197753813021769
#Rotation x (yaw)
0.06278809969454573
```

zongwave, 2019-04-02:
hi,
the units of positions is in millimeters, rotations is in degrees
Please refer to this page to understand how to get extrinsic parameters of a fisheye camera.
https://sites.google.com/site/scarabotix/ocamcalib-toolbox

Thanks,
Zong Wei

simpzan, 2019-04-03:
@zongwave seems like `Export Data` only export intrinsic parameters. so I need to modify the matlab code to export the `RRfin` to get the extrinsic parameters, right?

zongwave, 2019-04-03:
this tool export extrinsic parameters RRfin in terms of rotation matrix , you should convert the matrix into Euler angles roll/pitch/yaw

https://sites.google.com/site/scarabotix/ocamcalib-toolbox

RRfin | Contains the extrinsic parameters of every checkerboard with the respect to the camera reference frame. RRfin is a 3x3xM matrix, where M correspond to the respective image number. The first two columns of RRfin identify the ROTATION matrix (which is given by the column vectors Rx and Ry; Rz can be obtained as cross product between Rx and Ry). The third column of RRfin identifies the TRANSLATION vector of the checkerboard (i.e. [tx;ty;tz])
-- | --

simpzan, 2019-04-04:
@zongwave I just get the following `RRfin` matrix. 
```
   0.008067090116959  -0.001721306462260  -0.598260206191614
   0.002777823138339   0.009548147942833  -1.367026172935751
   0.005215913693398  -0.002422803113083  -3.522215947851525
```

according to the doc, 
- Rx is the first column, namely `(0.008067090116959, 0.002777823138339, 0.005215913693398)` 
- and Ry is the second column, namely `(-0.001721306462260, 0.009548147942833, -0.002422803113083)`, 
- and Rz = Rx cross product Ry, namely `(-0.00005653243414877, 0.00001056678510185, 0.00008180725482393)`.

now Rx, Ry, Rz vectors are all known, how to get the rotation matrix? I just googled a bit, the Rx is matrix of following form, not a vector. so I am still confused.
```
         | 1  0       0     |
    Rx = | 0  cos A  -sin A |
         | 0  sin A   cos A |
```

zongwave, 2019-04-11:
RRfin defines a rotation 3-vector, matlab function "rodrigues" can be used to calculate corresponding rotation matrix. 
details of rodrigues rotation formula at wiki:
https://en.wikipedia.org/wiki/Rodrigues%27_rotation_formula

  rodrigues	Transform rotation matrix into rotation vector and viceversa.
 		
 		Sintax:  [OUT]=rodrigues(IN)
  		If IN is a 3x3 rotation matrix then OUT is the
 		corresponding 3x1 rotation vector
  		if IN is a rotation 3-vector then OUT is the 
 		corresponding 3x3 rotation matrix

liss1810, 2019-05-30:
> this tool export extrinsic parameters RRfin in terms of rotation matrix , you should convert the matrix into Euler angles roll/pitch/yaw
>
> https://sites.google.com/site/scarabotix/ocamcalib-toolbox [...]

Hi, @zongwave , extrinsic parameters of a fisheye camera means the position and Euler angles of a fisheye camera in world(vehicle) reference frame?

wangqiang624731186, 2019-08-23:
> @zongwave I just get the following `RRfin` matrix.
>
> ```
>    0.008067090116959  -0.001721306462260  -0.598260206191614
>    0.002777823138339   0.009548147942833  -1.367026172935751 [...]

Have you solved the problem yet

wangqiang624731186, 2019-08-23:
> @zongwave I just get the following `RRfin` matrix.
>
> ```
>    0.008067090116959  -0.001721306462260  -0.598260206191614
>    0.002777823138339   0.009548147942833  -1.367026172935751 [...]

Have you solved the problem yet?

wangqiang624731186, 2019-08-23:
@simpzan Have you solved the problem yet?

closed 2022-01-13 by zongwave

shreshtashetty, 2023-06-12:
This might be a bit late, but just in case anyone working with libxcam moving forward has this doubt, following is the confirmation I have gotten from the OCamCalib forums:

```
The affine transform RRfin is [x y t]. Thus, to create a rotation matrix of RRfin for frame/image f do the following:

x = RRfin(1,:,frame);
y = RRfin(2,:,frame);
T = RRfin(3,:,frame);
z = cross(x,y);

R = [x y z]

```
And then with the R just created, the euler angles can be got as follows (in python3 code):

```
from scipy.spatial.transform import Rotation

r = Rotation.from_matrix(R)
angles = r.as_euler("zyx", degrees=True)
```

I hope this is right. If not, I hope the people maintaining this repo correct me.

sandeshchand87, 2023-11-22:
> hi, the units of positions is in millimeters, rotations is in degrees Please refer to this page to understand how to get extrinsic parameters of a fisheye camera. https://sites.google.com/site/scarabotix/ocamcalib-toolbox
> 
> Thanks, Zong Wei

This link is not available. It is not cleared to extract the extrinsic parameters from ocamcalib tool.