intel/libxcam #657: Calibration makes surround view stitching more distorted

Tamme, 2019-08-01:
Hi.

I calibrated my cams using matlab. Initially the projection is totally messed up:
[image]

After i change some signs of values in the intrinsic calibration inverse mapping it becomes normal, but the fisheye effect is actually increased:
[image]

After tinkering with pitch (set it to 15 degs) I can get it to be less distorted, bus my cameras are definitely not at this angle. 
[image]

Did you maybe use some other calibration setup, or have other ideas, what could be messed up?

Thank you!

closed 2022-01-13 by zongwave

mrp1234567, 2022-01-19:
@Tamme need some help on surround view with libxcam, can you please list the steps you did ?

Tamme, 2022-01-28:
Hi @mrp1234567. As mentioned, I used matlab ocam calib toolbox to get calibration values for the intrinsics. It has been some time, but I think then used [this example](https://github.com/intel/libxcam/blob/master/tests/test-image-stitching.cpp) as the executable to run the stitcher. And then I got some reasonable results with all zero extrinsics, I started modifying them, to make the result more correct.

Hope it helps.