intel/libxcam #658: Android compilation

yash4ever, 2019-08-07:
trying to compile libxcam under android while compiling got following issue 
make: *** No rule to make target 'external/libxcam/xcore/surview_fisheye_dewarp.cpp', needed by 'out/target/product/msm8909/obj/SHARED_LIBRARIES/libxcam_intermediates/xcore/surview_fisheye_dewarp.o'.  Stop.

zongwave, 2019-08-08:
Hi, we stop maintain Android build for some time.  We rename surview_fisheye_dewarp.cpp to libxcam/xcore/fisheye_dewarp.cpp. do you mind to change Android.mk by yourself and submit a pull request to fix the Android build issue?
Thanks.

yash4ever, 2019-08-08:
yeah sure i will do that but could you please tell me as its absolute from where i could get upstream source for android. i fixed that androdi.mk and after that i run into another issue any lead for androud would be really useful.

target thumb C++: libxcam <= external/libxcam/xcore/interface/stitcher.cpp
external/libxcam/xcore/interface/stitcher.cpp: In member function 'XCamReturn XCam::Stitcher::init_camera_info()':
external/libxcam/xcore/interface/stitcher.cpp:261:32: error: 'getenv' is not a member of 'std'
         std::string cfg_path = std::getenv (FISHEYE_CONFIG_ENV_VAR);

zongwave, 2019-08-08:
we upstream libxcam to Android AOSP two years ago, since then we did not update libxcam changes to AOSP. the libxcam source code in AOSP is out of date, you can find at here:
https://android.googlesource.com/platform/external/libxcam/+/refs/heads/master/

yash4ever, 2019-08-08:
i am trying to build libxcam for aosp could you please let me know which souce to use and how to compile it in aosp?

zongwave, 2019-08-08:
just make a full AOSP build, the source code in /externel/libxcam will be built. the output share library named as libxcam.so

yash4ever, 2019-08-08:
with the source present in https://android.googlesource.com/platform/external/libxcam/+/refs/heads/master/ 
?

yash4ever, 2019-08-21:
with above souce i am able to get
 "/obj/SHARED_LIBRARIES/libxcam_intermediates" 
but i am not getting any application binary.

zongwave, 2019-09-06:
libxcam built output shared libraries. please reference code in https://android.googlesource.com/platform/external/libxcam/+/refs/heads/master/tests/
to build Android application by yourself.

closed 2022-01-13 by zongwave