intel/libxcam #663: Input image

Naveen2195, 2019-08-23:
Hello,
Where can i get some sample test images in nv12 format for surround view and stiching?
I converted one mp4 video to yuv format and then renamed .yuv with .nv12 extension.
But it's throwing up this error

./test-surround-view --module gles --input /home/aiiec/naveen/opengl/libxcam-master/images/indoor.nv12 --output out.nv12
input0 file:		/home/aiiec/naveen/opengl/libxcam-master/images/indoor.nv12
output file:		out.nv12
input width:		1280
input height:		800
output width:		1920
output height:		640
topview width:		1280
topview height:		720
fisheye number:		4
resolution mode:	1080p4cams
blend pyr levels:	2
dewarp mode: 		bowl
scopic mode:		mono
scaling mode:		singleconst
feature match:		none
frame mode:		multiframe
save output:		true
save topview:		false
loop count:		1
XCAM INFO egl/egl_base.cpp:59: EGL version: 1.4
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Aborted (core dumped)

Kindly help.

liuyinhangx, 2019-08-23:
hi, How many inputs do you use, if 4, the test cmdline should be like this:  $ test-surround-view --module gles --output output.nv12 --input input0.nv12 --input input1.nv12 --input input2.nv12 --input input3.nv12 --fisheye-num 4 --res-mode 1080p4cams ......

Naveen2195, 2019-08-23:
Hello Liu,
I doubt my input images are causing the problem.
Tried what u said, this is what i got:
./test-surround-view --module gles --output output.nv12 --input 1.yuv --input 2.yuv --input 3.yuv --input 4.yuv --fisheye-num 4 --res-mode 1080p4cams 
input0 file:		1.yuv
input1 file:		2.yuv
input2 file:		3.yuv
input3 file:		4.yuv
output file:		output.nv12
input width:		1280
input height:		800
output width:		1920
output height:		640
topview width:		1280
topview height:		720
fisheye number:		4
resolution mode:	1080p4cams
blend pyr levels:	2
dewarp mode: 		bowl
scopic mode:		mono
scaling mode:		singleconst
feature match:		none
frame mode:		multiframe
save output:		true
save topview:		false
loop count:		1
XCAM INFO egl/egl_base.cpp:59: EGL version: 1.4
free(): invalid pointer
Aborted (core dumped)

  If i can get some sample images for surround view, it could help me( if the images are causing the problem here).

liuyinhangx, 2019-08-23:
@Naveen2195 , if the NV12 inputs are right, you can refer to the detailed test cmdline https://github.com/intel/libxcam/wiki/Tests#1-test-surround-view

Naveen2195, 2019-08-26:
@liuyinhangx : Thanks for the prompt response.
Could please provide NV12 inputs, i feel the my inputs are wrong.

Actually I have taken fisheyes images of PNG format, with help of FFMPEG tool I converted to .yuv extension...Then I renamed .yuv to .nv12 file extension names.

Kindly correct my above mentioned procedure...

Looking for your prompt reply.

liuyinhangx, 2019-08-26:
hi, @Naveen2195
1. You can use yuv viewer or player to verify the xxx.nv12 files.
2. For one frame(x4 inputs), you can try the two options on test-surround-view: --frame-mode single and --loop 100

Naveen2195, 2019-08-26:
Hello @liuyinhangx 
 I have tried that also. I cannot identify the problem from the error message on terminal, about whether it is caused by the images i gave or some other library/installation errors.

After reinstalling , I am getting a new error
./test-surround-view --input 1.nv12 --input 2.nv12 --input 3.nv12 --input 4.nv12 --output 5.nv12 --in-w 512 --in-h 512 --out-w 2048 --out-h 512  --frame-mode single  --loop 100
input0 file:		1.nv12
input1 file:		2.nv12
input2 file:		3.nv12
input3 file:		4.nv12
output file:		5.nv12
input width:		512
input height:		512
output width:		2048
output height:		512
topview width:		1280
topview height:		720
fisheye number:		4
resolution mode:	1080p4cams
blend pyr levels:	2
dewarp mode: 		bowl
scopic mode:		mono
scaling mode:		singleconst
feature match:		none
frame mode:		singleframe
save output:		true
save topview:		false
loop count:		100
test-surround-view: test-surround-view.cpp:77: void SVStream::set_module(SVModule): Assertion `module != SVModuleNone' failed.
Aborted (core dumped)

I am  new to image stiching and camera concepts. No idea about this error.

liuyinhangx, 2019-08-27:
hi @Naveen2195 
some information that may be helpful:
1. you lost some options, the cmdline should be like this:
  $ test-surround-view --module gles --output output.nv12 --input input0.nv12 --input input1.nv12 --input input2.nv12 --input input3.nv12 --in-w 1280 --in-h 800 --out-w 1920 --out-h 640 --topview-w 1280 --topview-h 720 --fisheye-num 4 --res-mode 1080p4cams --dewarp-mode bowl --scale-mode dualconst --frame-mode single --fm-mode cluster --save true --save-topview false --loop 100
2. for detailed test, you can refer to [test-surround-view](https://github.com/intel/libxcam/wiki/Tests#1-test-surround-view)
3. you can use the follow cmdline to get help information:
  $ test-surround-view --help
4. if --dewarp-mode is bowl, you shoud do the fisheye calibration first, you can refer to https://github.com/intel/libxcam/issues/497

Naveen2195, 2019-08-27:
Hello Liu, 
I tried running with above parameters except   --fm-mode cluster ( shows error: _XCAM ERROR test-surround-view.cpp:655: surround view unsupported feature match mode: cluster_ )
Please note i used custom resolutions for -w and -h as per my image.
After running i got this error msg;

XCAM INFO egl/egl_base.cpp:59: EGL version: 1.4
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Aborted (core dumped)

 The images are calibrated I believe (Will try again).

I don't have xcam folder in my etc or anywhere. So I am afraid  
_export FISHEYE_CONFIG_PATH=/etc/xcam/calibration_   is causing problem?
No idea why it's not created after building everything.

Naveen2195, 2019-08-27:
Hello Liu,
For calibration, in what format we should give the intrinsic parameters?
Currently it's in text format. 
Also Can you please suggest how to give intrinsic and extrinsic parameter values?

liuyinhangx, 2019-08-27:
Some dependent libraries may not be installed, please refer to [Build](https://github.com/intel/libxcam/wiki/Build), and verify OpenCV and Mesa3D are installed correctly.

liuyinhangx, 2019-08-27:
hi @Naveen2195 
Text format(xxxx.txt) is ok, you can refer to https://github.com/intel/libxcam/issues/497
From [test-surround-view](https://github.com/intel/libxcam/wiki/Tests#1-test-surround-view):
The calibration files(intrinsic paramters and extrinsic parameters) should be stored in "FISHEYE_CONFIG_PATH" directory, and then export the path:
```Shell
        export FISHEYE_CONFIG_PATH=/etc/xcam/calibration
```
You only need to export the path of calibration files(xxxx.txt).

jeongseokhwan, 2019-08-30:
hi~i have a question..
i downloaded the libxcam_app and libxcam-master two tar files and i tested.
i used the 4 cameras(of course i got the intrinsic parameter from ocam calibration and extrinsic parameter from opencv), but result is very strange.. i connected camera front,right,rear,left
output image is very not good. i think even if intrinsic, extrinsic parameter is wrong, in the output image, 4 cameras input are displayed regular..but it is not..
anyone know why? or reason? please help me..
thanks
[image]

liuyinhangx, 2019-08-30:
hi @jeongseokhwan 
You can refer to https://github.com/intel/libxcam/issues/497, https://github.com/intel/libxcam/issues/616 and https://github.com/intel/libxcam/issues/650

jeongseokhwan, 2019-09-18:
hi~i have a question..
when i tested with 960 x 540 size image...it works..but i tested with 1280 x 720 size image.
it doesn't work..i also modified input, output, topview size in .cpp file(960 x 540 -> 1280 x 720)
free(): invalid pointer : 0x0000000000462900
aborted(core dumped)
anyone know??

closed 2022-01-13 by zongwave