intel/libxcam #669: share input image

wangqiang624731186, 2019-09-03:
this is my test image
branch: 1.2.2
QQ:624731186
email:624731186@qq.com
[test_image.zip](https://github.com/intel/libxcam/files/3567678/test_image.zip)

[image]

wangqiang624731186, 2019-09-03:
我现在想把最新的版本移植到android x86上，欢迎指导

zongwave, 2019-09-03:
please get source from Android AOSP: https://android.googlesource.com/platform/external/libxcam/

git clone https://android.googlesource.com/platform/external/libxcam

jeongseokhwan, 2019-09-06:
in test-surround-view.cpp, there are no input0, input1, input2, input3 in usage. so when i input --input0 or --input1, "unrecognized option '--input1' message is printed..how did you set the option? or source code?

liuyinhangx, 2019-09-06:
@jeongseokhwan 
After 1.2.2 release, we simplify the option name of --inputX to support different amounts of inputs, you can use "--input input0.nv12 --input input1.nv12 --input input2.nv12 --input input3.nv12", please refer the commit log of [test-surround-view: simplify the option name of --inputX](https://github.com/intel/libxcam/commit/2e4eb177dd63ad438b8b6e4a8ef21ec55a69064b)

jeongseokhwan, 2019-09-06:
@liuyinhangx 
thanks~i solved..thanks a lot

jeongseokhwan, 2019-09-30:
how to get intrinsic and extrinsic parameter?

manchengfenxu, 2020-07-15:
> how to get intrinsic and extrinsic parameter?

I think you can use this matlab tool, [https://www.mathworks.com/help/vision/ug/fisheye-calibration-basics.html]

closed 2022-01-13 by zongwave

lekhashreehj, 2023-05-31:
XCAM ERROR test-surround-view.cpp:865: surround view unsupported feature match mode: capi

lekhashreehj, 2023-05-31:
I am getting this error when I runned ./test-surround-view --module soft --input ./front.yuv --input ./right.yuv --input ./rear.yuv --input ./left.yuv --output output.nv12 --in-w 960 --in-h 540 --out-w 960 --out-h 540 --topview-w 960 --topview-h 540 --scale-mode dualcurve --fm-mode capi --frame-mode multi --save true --save-topview true --loop 1

Please help me to debug the error

dreamerfar, 2023-05-31:
please use "default" or "cluster" fm-mode. "capi" is to abstract C api to application such as ffmpeg.

lekhashreehj, 2023-06-01:
XCAM ERROR test-surround-view.cpp:865: surround view unsupported feature match mode: default

I am getting this error in both default and cluster.
> please use "default" or "cluster" fm-mode. "capi" is to abstract C api to application such as ffmpeg.

dreamerfar, 2023-06-01:
You should build/install & enable opencv first.

https://github.com/intel/libxcam/blob/master/tests/test-surround-view.cpp#LL852C1-L869C19

        case 'F':
            XCAM_ASSERT (optarg);
            if (!strcasecmp (optarg, "none"))
                fm_mode = FMNone;
#if HAVE_OPENCV
            else if (!strcasecmp (optarg, "default"))
                fm_mode = FMDefault;
            else if (!strcasecmp (optarg, "cluster"))
                fm_mode = FMCluster;
            else if (!strcasecmp (optarg, "capi"))
                fm_mode = FMCapi;
#endif
            else {
                XCAM_LOG_ERROR ("surround view unsupported feature match mode: %s", optarg);
                usage (argv[0]);
                return -1;
            }
            break;

CPFelix, 2023-06-16:
@wangqiang624731186 Thanks for your images.
Have you try to build 3D view using Vulkan module?
I used yout image and run:
_./test-surround-view --module vulkan --input /home/chenpengfei/AVM/test_image/front.yuv --input /home/chenpengfei/AVM/test_image/right.yuv --input /home/chenpengfei/AVM/test_image/rear.yuv --input /home/chenpengfei/AVM/test_image/left.yuv --output /home/chenpengfei/AVM/test_image/fuxian/output.nv12 --in-w 960 --in-h 540 --out-w 960 --out-h 540 --topview-w 960 --topview-h 540 --fisheye-num 4 --dewarp-mode bowl --scale-mode singleconst --frame-mode multi --fm-mode default --save true --save-topview true --loop 1_

get error:
_test-surround-view: interface/stitcher.cpp:601: XCamReturn XCam::Stitcher::update_copy_areas(): Assertion `left.in_area.width > 0' failed._

Adarine99, 2023-11-23:
> this is my test image branch: 1.2.2 QQ:624731186 email:[624731186@qq.com](mailto:624731186@qq.com) Download: [test_image.zip](https://github.com/intel/libxcam/files/3567678/test_image.zip)
> 
> export FISHEYE_CONFIG_PATH=./calib_params
> 
> ./test-surround-view --module soft --input0 ./front.yuv --input1 ./right.yuv --input2 ./rear.yuv --input3 ./left.yuv --output output.nv12 --in-w 960 --in-h 540 --out-w 960 --out-h 540 --topview-w 960 --topview-h 540 --scale-mode dualcurve --fm-mode capi --frame-mode multi --save true --save-topview true --loop 1
> 
> ffplay -f rawvideo -video_size 960x540 topview_output.nv12
> 
> [image]
> 
> [image]
你好大佬，想请问一下你的相机外参是怎么计算出来的？
感谢感谢~