intel/libxcam #683: Centre of video frame had been mess up

kchoonhw, 2019-10-03:
I am using this libxcam library and stitching for 4 sample input video frame to simulate the autonomous of bowl view feature. however, in the center of the car the image was mess up. May i know how can i fix it? 

[image]

jeongseokhwan, 2019-10-11:
hi~ i also have similar problem...
how did you get intrinsic parameter and extrinsic parameter? i got intrinsic parameter from ocam calibration and extrinsic parameter from real value(i set the camera by my self and measure the x, y, z)
[image]

thanks~

zongwave, 2019-10-11:
@kchoonhw you posted a topview image, which is a view form the top direction. you should know the horizontal FOV of input images is limited, no ground image captured under the car at all.
the application developer usually need to rend a car to cover the black hole in the center of topview.

closed 2022-01-13 by zongwave