intel/libxcam #698: Calbration for 2-fisheye equirectangular stitching

envyen, 2020-01-28:
Hello team

For the dual-fisheye to equirectangular stitching, what is the recommended calibration scheme/tool used? How is the calibration parameters used in the library

Kindly share the details/steps for this

zongwave, 2020-03-26:
to remap fisheye camera image into equirectangular image we use some intrinsic and extrinsic parameters:
principal axis (points)
radius
fov
rotation

envyen, 2020-06-12:
Hi thanks for the details
Is there any openly available tools/source to achieve this from input images ?
Any pointers to estimate the parameters would be helpful 

> principal axis (points)
> radius
> fov
> rotation

zongwave, 2020-06-12:
Principal axis and rotation can be acquired through OpenCV fisheye calibration tool.

Fov can be acquired from camera manufacture

Radius is in pixel unit, you can estimate the value from the fisheye image.

envyen, 2021-02-19:
Thanks Zong Wei, the information helped

closed 2021-02-19 by envyen