intel/libxcam #699: Processed 0 Buffers

vijay73anand, 2020-02-05:
Hello,

I am getting **XCAM INFO test-cl-image.cpp:542: processed 0 buffers successfully**, wehn trying to run 

**Command:** 
./test-cl-image -t gauss -f BA10   -i img1.raw  -o o12.raw -k kernel_bayer_basic.cl.bin

**Message:** 

XCAM WARNING cl_device.cpp:180: cl get device info(CL_DEVICE_IMAGE_PITCH_ALIGNMENT) failed
XCAM INFO cl_device.cpp:134: cl get device info,
	max_compute_unit:28	max_work_item_dims:3	max_work_item_sizes:{1024, 1024, 64}	max_work_group_size:1024	image_pitch_alignment:4
XCAM DEBUG cl_device.cpp:54: CL device constructed
XCAM DEBUG cl_context.cpp:95: CLContext constructed
XCAM DEBUG cl_context.cpp:700: CLCommandQueue constructed
XCAM INFO test-cl-image.cpp:542: processed 0 buffers successfully

Please resolve it as soon as possible

Thank you in advance

vijay73anand, 2020-02-06:
Command:
./test-cl-image -t gauss -f BA10 -i img1.raw -o o12.raw -k kernel_bayer_basic.cl.bin

Message:
I think it is because while pre-executing kernels in cl_image_handler.cpp, it fails with following error message:

XCAM WARNING cl_context.cpp:433: create cl image failed with -40:CL_INVALID_IMAGE_SIZE
XCAM ERROR cl_memory.cpp:597: CLImage2D create image 2d failed, image desc width:1920 height:1080 row_pitch:7680 slice_pitch:0 array_size:0
XCAM ERROR cl_utils.cpp:121: convert to climage failed, image desc width:1920 height:1080 row_pitch:7680 slice_pitch:0 array_size:0 size:0
XCAM WARNING cl_context.cpp:433: create cl image failed with -40:CL_INVALID_IMAGE_SIZE
XCAM ERROR cl_memory.cpp:597: CLImage2D create image 2d failed, image desc width:480 height:1080 row_pitch:7680 slice_pitch:0 array_size:0
Thread 1 "test-cl-image" received signal SIGSEGV, Segmentation fault.dth:480 height:1080 row_pitch:7680 slice_pitch:0 array_size:0 size:0
0x00007ffff78c877f in XCam::CLGaussImageKernel::prepare_arguments (this=0x555555b24490, args=empty std::__cxx11::list, work_size=...) at cl_gauss_handler.cpp:150

Please resolve it as soon as possible

Thank you in advance

zongwave, 2020-02-07:
Hi I just run the test successfully. seems you got some issues on your OpenCL driver.
Please refer to the OpenCL installation guide to install CL driver correctly.
https://github.com/intel/libxcam/wiki/Build

$ ./test-cl-image -t gauss -f BA10 -i  ba10-bayer-raw.raw -o output.raw

XCAM INFO cl_device.cpp:137: cl get device info,
        max_compute_unit:24     max_work_item_dims:3    max_work_item_sizes:{256, 256, 256}     max_work_group_size:256 image_pitch_alignment:4
XCAM DEBUG cl_device.cpp:57: CL device constructed
XCAM DEBUG cl_context.cpp:101: CLContext constructed
XCAM DEBUG cl_context.cpp:706: CLCommandQueue constructed
XCAM INFO test-cl-image.cpp:542: processed 1 buffers successfully

vijay73anand, 2020-02-10:
Hello,
Thank you for the quick reply. I am checked my opencl installation it all seems fine, so could you please end me the command for getting the kernel  kernel_bayer_basic.cl.bin. As i think i am going wrong there.

Thank you in advance

vijay73anand, 2020-02-10:
This is the command i used:

./test-binary-kernel --src-kernel ../shaders/cl/kernel_gauss.cl --bin-kernel kernel_gauss.cl.bin --kernel-name kernel_gauss

zongwave, 2020-02-11:
you can use this command to compile and execute cl kernel from source:
$ ./test-cl-image -t gauss -f BA10 -i ba10-bayer-raw.raw -o output.raw

or you can compile cl kernel with command:
$ ./test-binary-kernel --src-kernel ../shaders/cl/kernel_gauss.cl --bin-kernel kernel_gauss.cl.bin --kernel-name kernel_gauss
then execute the cl kernel binary file:
$ ./test-cl-image -t gauss -f BA10 -i ../../../test_video/ba10-bayer-raw.raw -o output.raw  -k kernel_bayer_basic.cl.bin
output:
XCAM INFO cl_device.cpp:137: cl get device info,
        max_compute_unit:24     max_work_item_dims:3    max_work_item_sizes:{256, 256, 256}     max_work_group_size:256 image_pitch_alignment:4
XCAM DEBUG cl_device.cpp:57: CL device constructed
XCAM DEBUG cl_context.cpp:101: CLContext constructed
XCAM DEBUG cl_context.cpp:706: CLCommandQueue constructed
XCAM INFO test-cl-image.cpp:542: processed 1 buffers successfully

closed 2022-01-13 by zongwave