intel/libxcam #729: CONFIGURE FFMPEG

logindiamant, 2020-06-04:
Dear colleagues, please tell us, how to configure libxcam in ffmpeg?

I try to build ffmpeg by sources, step by step
https://trac.ffmpeg.org/wiki/CompilationGuide/Ubuntu
...
When I try to compile ffmpeg with
cd ~/ffmpeg_sources && \
wget -O ffmpeg-snapshot.tar.bz2 https://ffmpeg.org/releases/ffmpeg-snapshot.tar.bz2 && \
tar xjvf ffmpeg-snapshot.tar.bz2 && \
cd ffmpeg && \
PATH="$HOME/bin:$PATH" PKG_CONFIG_PATH="$HOME/ffmpeg_build/lib/pkgconfig" ./configure \
  --prefix="$HOME/ffmpeg_build" \
  --pkg-config-flags="--static" \
  --extra-cflags="-I$HOME/ffmpeg_build/include" \
  --extra-ldflags="-L$HOME/ffmpeg_build/lib" \
  --extra-libs="-lpthread -lm" \
  --bindir="$HOME/bin" \
  --enable-gpl \
  --enable-gnutls \
  --enable-libaom \
  --enable-libass \
  --enable-libfdk-aac \
  --enable-libfreetype \
  --enable-libmp3lame \
  --enable-libopus \
  --enable-libvorbis \
  --enable-libvpx \
  --enable-libx264 \
  --enable-libx265 \
  **--enable-libxcam** \
  --enable-nonfree && \
PATH="$HOME/bin:$PATH" make && \
make install && \
hash -r
...
I see error
**Unknown option "--enable-libxcam".**

zongwave, 2020-06-04:
We implemented a libXCam video filter for FFMPEG, but we have not submitted the video filter to FFMpeg official repo…

https://trac.ffmpeg.org/wiki/CompilationGuide/Ubuntu
...

logindiamant, 2020-06-04:
@zongwave, please, give us unofficial FFMPEG build with libXCam…
email: gleb.derzkij@gmail.com

zongwave, 2020-06-05:
Hi, we have to go through Intel Open Source process before publish this FFMPEG video filter. I’ll inform you then.

Thanks,
Zong Wei

email: gleb.derzkij@gmail.com<mailto:gleb.derzkij@gmail.com>

tsgao, 2021-11-09:
I’m also wondering how to configure libxcam for ffmpeg on Ubuntu. Has this went through the Intel open source process yet?

zongwave, 2021-11-11:
Please refer to this patch we submitted to FFMpeg patchwork
https://patchwork.ffmpeg.org/project/ffmpeg/list/?series=1904

closed 2022-01-13 by zongwave