intel/libxcam #731: Building with OpenCV

aliabbas2810, 2020-06-09:
Hi,
I cannot specify the OpenCV version with building libxcam. Opencv was install while installing OpenVino and its version is 4.3.0. If I do not specify the version then i get an error, if I give the version then i get the following warning and linxcam is not configured with opencv

I am getting the following warning.
configure: WARNING: unrecognized options: --enable-opencv-4.3.0

     libxcam configuration summary
     version                    : 1.4.0
     enable debug               : yes
     enable profiling           : yes
     enable drm lib             : yes
     build GStreamer plugin     : yes
     build aiq analyzer         : no
     use local aiq              : no
     use local atomisp          : no
     enable OpenCV              : no
     enable OpenCL              : yes
     enable GLES                : yes
     enable Vulkan              : no
     enable AVX512 instruction  : no
     enable OSG render          : yes
     enable DNN                 : yes
     enable 3a lib              : yes
     enable smart analysis lib  : yes
     enable dvs                 : no
     enable libxcam-capi lib    : yes
     enable json parser         : yes

aliabbas2810, 2020-06-09:
i have tried with --enable-opencv_4.3.0 and --enable-opencv 4.3.0. I am getting the same error

zongwave, 2020-06-09:
Hi, to build libXCam with “—enable-opencv” option, it requires OpenCV<http://opencv.org/> versions [3.0.0 - 4.0.0), cause OpenCV APIs changed a lot after version 4.0.0. we need some extra effort to enable OpenCV about 4.0.0.

aliabbas2810, 2020-06-11:
I installed and tested opencv 4.0.0 and its still the same problem.

zongwave, 2020-06-11:
Please install opencv 3.x

zongwave, 2020-06-12:
libxcam using "pkg-config" to get opencv "inc & library" path while compile, you can "make" and "sudo make install" opencv 3.x at folder "/usr/local"

$ pkg-config --modversion opencv
3.4.3
$ pkg-config --libs opencv
-L/usr/local/lib -lopencv_stitching -lopencv_superres -lopencv_videostab

if you build OpenVino with instruction (https://github.com/openvinotoolkit/openvino/blob/master/build-instruction.md#build-on-linux-systems), OpenCV 4.3 will be installed at "openvino/inference-engine/temp/opencv_4.3.0_ubuntu16/opencv/lib"

You can maintain two versions at the same time, AutoMake will output warning while compile libxcam, just ignore the warning, all functions should work well.

libtool: link: g++ -fPIC -Wall -fstack-protector -g -DDEBUG -std=c++0x -I../xcore -I../modules -I/usr/local/include/opencv -I/usr/local/include -I/home/wzong/Workspace/libs/cv/openvino/inference-engine/include -g -O3 -o .libs/test-dnn-inference test_dnn_inference-test-dnn-inference.o  ../xcore/.libs/libxcam_core.so ../modules/ocv/.libs/libxcam_ocv.a -L/usr/local/lib -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_dnn_objdetect -lopencv_dpm -lopencv_face -lopencv_photo -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_hfs -lopencv_img_hash -lopencv_line_descriptor -lopencv_optflow -lopencv_reg -lopencv_rgbd -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_surface_matching -lopencv_tracking -lopencv_datasets -lopencv_text -lopencv_dnn -lopencv_plot -lopencv_xfeatures2d -lopencv_shape -lopencv_video -lopencv_ml -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_highgui -lopencv_videoio -lopencv_flann -lopencv_xobjdetect -lopencv_imgcodecs -lopencv_objdetect -lopencv_xphoto -lopencv_imgproc -lopencv_core /home/wzong/Workspace/xcam/zongwave/libxcam/xcore/.libs/libxcam_core.so -ldl ../modules/soft/.libs/libxcam_soft.so ../modules/dnn/inference/.libs/libxcam_dnn.so -pthread

/usr/bin/ld: warning: libopencv_imgcodecs.so.4.3, needed by //home/wzong/Workspace/libs/cv/openvino/bin/intel64/Release/lib/libformat_reader.so, may conflict with libopencv_imgcodecs.so.3.4

aliabbas2810, 2020-06-24:
hi
I installed opencv-3.4.8 at /user/local/opencv-3.4.8 and tested it. How can i specify the path of opencv while building linxcam.
is it possible or do we have to install opencv at /user/local
My setup depends on using the current version of opencv so i need to use the current version and also opencv-3.x.x for libxcam.

thanks

closed 2020-06-26 by aliabbas2810

StefanBruens, 2020-07-23:
Actually, OpenCV did not change with version 4.0, but with version 2.0 when the C++ API was introduced. OpenCV 4.1. finally removed the old C API which was deprecated for years.

The only remaining user of the old C API is the optical flow feature matcher, used by the soft_stitcher. As far as I can see there already is a functional identical feature matcher using the C++ API.

Any reason why soft_stitcher does not just use `CvFeatureMatch` instead of `CVCapiFeatureMatch`?

zongwave, 2020-07-24:
We implemented CVCapiFeatureMatch for Android build, since Android only include C version of OpenCV in it's external component.

StefanBruens, 2020-07-24:
The latest SDK definitely exports the C++ ABI from its libs:
```
nm -DC libopencv_java4.so  | grep cv::goodFeatures
00000000006d5260 T cv::goodFeaturesToTrack(cv::_InputArray const&, cv::_OutputArray const&, int, double, double, cv::_InputArray const&, int, bool, double)
00000000006d3110 T cv::goodFeaturesToTrack(cv::_InputArray const&, cv::_OutputArray const&, int, double, double, cv::_InputArray const&, int, int, bool, double)
```