intel/libxcam #732: fatal error: inference_engine.hpp: No such file or directory

aliabbas2810, 2020-06-11:
Hi,
I have install intel openvino and tested it. Now i am trying to build libxcam but i am getting fatal error. openvino is installed on /opt/intel/openvino/
inference_engine.hpp is available on /opt/intel/openvino/inference_engine/include but i am getting the error.

following is the error message.
```
make[3]: Entering directory '/home/ali/Desktop/SoftwareDev/ThirdParties/libxcam/modules/dnn'
Making all in inference
make[4]: Entering directory '/home/ali/Desktop/SoftwareDev/ThirdParties/libxcam/modules/dnn/inference'
/bin/bash ../../../libtool  --tag=CXX --tag=disable-static  --mode=compile g++ -DHAVE_CONFIG_H -I. -I../../..    -fPIC -Wall -fstack-protector -g -DDEBUG -std=c++0x -I../../../xcore -I../../../modules -I/opt/intel/openvino/inference_engin/include/include -I/opt/intel/openvino/inference_engin/include/src/extension -I/opt/intel/openvino/inference_engin/include/samples/common/format_reader   -g -O2 -MT libxcam_dnn_la-dnn_inference_engine.lo -MD -MP -MF .deps/libxcam_dnn_la-dnn_inference_engine.Tpo -c -o libxcam_dnn_la-dnn_inference_engine.lo `test -f 'dnn_inference_engine.cpp' || echo './'`dnn_inference_engine.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../../.. -fPIC -Wall -fstack-protector -g -DDEBUG -std=c++0x -I../../../xcore -I../../../modules -I/opt/intel/openvino/inference_engin/include/include -I/opt/intel/openvino/inference_engin/include/src/extension -I/opt/intel/openvino/inference_engin/include/samples/common/format_reader -g -O2 -MT libxcam_dnn_la-dnn_inference_engine.lo -MD -MP -MF .deps/libxcam_dnn_la-dnn_inference_engine.Tpo -c dnn_inference_engine.cpp  -fPIC -DPIC -o .libs/libxcam_dnn_la-dnn_inference_engine.o
In file included from dnn_inference_engine.cpp:21:0:
dnn_inference_engine.h:28:10: fatal error: inference_engine.hpp: No such file or directory
 #include <inference_engine.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:530: recipe for target 'libxcam_dnn_la-dnn_inference_engine.lo' failed
make[4]: *** [libxcam_dnn_la-dnn_inference_engine.lo] Error 1
make[4]: Leaving directory '/home/ali/Desktop/SoftwareDev/ThirdParties/libxcam/modules/dnn/inference'
Makefile:387: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/home/ali/Desktop/SoftwareDev/ThirdParties/libxcam/modules/dnn'
Makefile:393: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/ali/Desktop/SoftwareDev/ThirdParties/libxcam/modules'
Makefile:436: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/ali/Desktop/SoftwareDev/ThirdParties/libxcam'
Makefile:368: recipe for target 'all' failed
make: *** [all] Error 2
```

zongwave, 2020-06-11:
Hi, libxcam implemented DNN module last year, since then OpenVino changed APIs dynamically, it cannot build libxcam with latest OpenVino library. I’m updating libxcam dnn module to complied with OpenVino changes, but not finished yet. Could you please remove “—enable-dnn” in configuration command to skip build “libxcam/modules/dnn” ?

zongwave, 2020-06-12:
https://github.com/intel/libxcam/pull/733

please check the commit message how to build libxcam with --enable-dnn

aliabbas2810, 2020-06-24:
Thanks. I am able to build it with dnn option.

closed 2020-06-30 by zongwave