intel/libxcam #735: test-image-stitching with IP cameras

aliabbas2810, 2020-06-29:
Hi,
I have two IP cameras. I have configured the ip and I am able to connect to the cameras through a web plugin. I am also able to get the input from the cameras in my Opencv code. Could you please look into the command line arguments for test-image-stitching i,e. 
```
./test-image-stitching --input rtsp://Admin:indigo1234@10.5.1.108 --input rtsp://Admin:indigo1234@10.5.1.106 --output output.mp4 --input-w 1920 --input-h 1080 --output-w 1920 --output-h 960 --dewarp-mode sphere --scale-mode local --enable-fisheyemap --res-mode 1080p2cams --framerate 30.0 --fm true --save true --save-topview false --save-freeview false --loop 1
input0 file:		rtsp://Admin:indigo1234@10.5.1.108
input1 file:		rtsp://Admin:indigo1234@10.5.1.106
output file:		output.mp4
input width:		1920
input height:		1080
output width:		1920
output height:		960
resolution mode:	1080p2cams
fisheye dewarp mode: 	sphere
scale mode:		local
seam mask:		false
fisheye map:		true
shading correction:	false
feature match:		true
fisheye number:		2
save file:		true
save topview file:	false
save freeview file:	false
framerate:		30.000
loop count:		1
```
I am getting the following error
```
XCAM ERROR test_stream.h:187: stream(rtsp://Admin:indigo1234@10.5.1.108) open failed
XCAM ERROR test-image-stitching.cpp:527: open input file(rtsp://Admin:indigo1234@10.5.1.108) failed

```
Regards,
Ali

zongwave, 2020-06-30:
Hi, the test-image-stitching does not handle input stream from camera URL . If you read the source code, you will find class Stream is using class FileHandle to manipulate image data, which defines FILE open, read, write operations on FILE.

https://github.com/intel/libxcam/blob/master/tests/test_stream.h#L84
https://github.com/intel/libxcam/blob/master/xcore/file_handle.h#L29

I suggest you to save video data into files, convert into YUV420 format, then do the testing.

aliabbas2810, 2020-06-30:
Thanks. So I have to make changes or write my one code to handle live cameras. I also need to know about the input format i.e YUV420, can we use other input formats or the does the algorithms being used

zongwave, 2020-07-01:
test-image-stitching use OpenCL computing, it only supports NV12 input format.
test-surround-view accelerated by CPU AVX instruction with "--module soft" option, it supports both NV12 and YUV420p format "--in-format".

closed 2022-01-13 by zongwave