intel/libxcam #786: BowlDataConfig parameters in surround view application

ycchen810720, 2021-11-08:
Hi
I'm working on applications related to car surround view using four fisheye lenses.
There is a BowlDataConfig parameter, which is closely related to the result of the entire projection and stitching
In test-surround-view project, the parameters are:
bowl.a = 6060.0f;
bowl.b = 4388.0f;
bowl.c = 3003.4f;
bowl.angle_start = 0.0f;
bowl.angle_end = 360.0f;
bowl.center_z = 1500.0f;
bowl.wall_height = 1800.0f;
bowl.ground_length = 3000.0f;
I have noticed that there are corresponding parameter explanations
But I still don’t know how to set the parameters according to the actual application
It seems to be related to the angle of view of the lens and the camera location(extrinsic parameters)?

zongwave, 2022-01-13:
this structure is defined at: https://github.com/intel/libxcam/blob/master/xcore/interface/data_types.h#L187
it is construct a bowl model that used to rendering surround view texture.

the intrinsic and extrinsic parameters are defined at: https://github.com/intel/libxcam/blob/master/xcore/interface/data_types.h#L128
you need to use camera calibration tool to generate camera calibration files, please check calibration related discussion here: https://github.com/intel/libxcam/issues/793

ycchen810720, 2022-01-13:
Thank you very much for answering my question.
My problem is how did the above set of numbers get.
Here is my inference:
1. Assume car_width=1600, half_car_width=800
                  car_length=4500, half_car_height=2250
                  car_height=1500
2.Set bowl.center_z =car_height
3.Assume ground_length =3000
4.Assume bowl.c=3003.4
5.Set bowl.a=(ground_length+half_car_height)/sqrt(1-pow(bowl.center_z,2)/pow(bowl.c,2))=6059.89≈6060
         bowl.b=(ground_length+half_car_width)/sqrt(1-pow(bowl.center_z,2)/pow(bowl.c,2))=4386.20≈4388
6.Assume bowl.wall_height=1800 (human length?)

Is that correct or there is a clearer formula?
The numbers in bowl.c seem to be calculated rather than an assumption.

Adarine99, 2022-07-19:
same question,I know that:
/*
 * Ellipsoid model
 *  x^2 / a^2 + y^2 / b^2 + (z-center_z)^2 / c^2 = 1
 * ground : z = 0
 * x_axis : front direction
 * y_axis : left direction
 * z_axis : up direction
 * wall_height : bowl height inside of view
 * ground_length: left direction distance from ellipsoid bottom edge to nearest side of the car in the view
 */But pWhat does the parameter (x  y  z)mean