intel/libxcam #797: If surround view is working with libxcam, can anyone please post output image ?

mrp1234567, 2022-01-19:
Hi Everyone,

If anyone is successfully able to generate surround view with 4 cameras, can you please post your camera models, fov, configurations done, calibration files, input images/videos and output image/video ?

zongwave, 2022-01-20:
Please check the last comments I posted for issue #793 

https://github.com/intel/libxcam/issues/793#issuecomment-1017201234

fanDongHust, 2022-05-05:
T_FG:

           1            0            0 -1.64277e-15

           0    -0.707107    -0.707107     -24.5389

           0     0.707107    -0.707107      13.4022

           0            0            0            1

T_LG:

          0           1 5.55112e-17 1.77636e-15

   0.707107           0   -0.707107    -11.9349

  -0.707107 5.55112e-17   -0.707107    0.798176

          0           0           0           1

T_BG:

          -1 -1.22465e-16  6.16298e-33  1.64277e-15

-8.65956e-17     0.707107    -0.707107     -24.5389

 8.65956e-17    -0.707107    -0.707107      13.4022

           0            0            0            1

T_RG:

           0           -1 -5.55112e-17 -2.66454e-15

   -0.707107            0    -0.707107     -11.9349

    0.707107  5.55112e-17    -0.707107     0.798176

           0            0            0            1

K_F:

320   0 640

  0 320 360

  0   0   1

D_F:

  0.0830922

  0.0111213

-0.00858726

0.000854219

[image]

left-top: front camera;
right-top: left camera;
left-bottom: rear camera;(back camera)
right-bottom: right camera;