intel/libxcam #816: How to add 2 inputs to gstreamer's xcamfilter plugin for stitch

derekoutdu, 2022-12-28:
Hi, 

Can anyone tell me how to pass two input files for stitching two source using gstreamer plugin xcamfilter?
The provided test command is not explained for adding multiple filesrc
given command: gst-launch-1.0 filesrc location=input.nv12 ! videoparse width=1920 height=1080 format=nv12 ! xcamfilter copy-mode=1 enable-stitch=true stitch-scale=local stitch-fisheye-map=true stitch-lsc=true stitch-fm-ocl=true stitch-res-mode=1080p ! video/x-raw, foramt=NV12, width=1920, height=960 ! queue ! vaapiencode_h264 rate-control=cbr ! qtmux ! filesink location=test.mp4

as per this given command, the file source is only input.nv12. What is it stitching if there's only 1 input?

brmarkus, 2023-01-09:
Is this topic still open?

derekoutdu, 2023-01-10:
I haven't gotten any response on the issue I'm facing in. So yes, the topic is still open.

zongwave, 2023-01-10:
sorry libxcam gstream plugin "xcamfilter" does not support input multiple images.
we have ffmpeg filter that support input 3 or 4 images stitching.

zongwave, 2023-01-10:
> Hi,
>
> Can anyone tell me how to pass two input files for stitching two source using gstreamer plugin xcamfilter? The provided test command is not explained for adding multiple filesrc given command: [...]

the 360 camera we used combined the front and rear view images into one frame, please check the 2 cameras stitching process flow at https://github.com/intel/libxcam/wiki#libxcam-stitch-processing-flow

closed 2023-01-14 by derekoutdu