intel/libxcam #822: 'dump_dma_video_buf' was not declared in this scope

Douaty, 2023-05-06:
1、test-surround-view.cpp:507:21：error ’dump_dma_video_buf‘ was not declared in this scope
2、test-surround-view.cpp:1064:64：error ‘XCAM_GL_RESERVED_BUF_COUNT’ was not declared in this scope

sjuxax, 2023-05-11:
Happens when built without `--enable-gles`.

Douaty, 2023-06-09:
Hi ，May I ask the following questions，Thinks！
./test-surround-view --module gles --input test/01.nv12 --input test/02.nv12 --output test/output.yuv --in-w 640 --in-h 360 --out-w 1120 --out-h 480 --topview-w 640 --topview-h 360 --in-format yuv --fisheye-num 2 --cam-model cama2c1080p --blend-pyr-levels 2 --dewarp-mode sphere --scopic-mode mono --scale-mode dualconst --frame-mode single --loop 1
input0 file:		test/01.nv12
input1 file:		test/02.nv12
camera model:		cama2c1080p
fisheye number:		2
stitch module:		GLES
enable DMA buffer input/output:		false
device node:		Not specified, use default model
output file:		test/output.yuv
input width:		640
input height:		360
output width:		1120
output height:		480
topview width:		640
topview height:		360
cubemap width:		1280
cubemap height:		720
input format:		yuv
blend pyr levels:	2
dewarp mode: 		sphere
scopic mode:		mono
scaling mode:		dualconst
feature match:		none
frame mode:		singleframe
save output:		true
save topview:		false
save cubemap:		false
loop count:		1
repeat count:		1
XCAM ERROR gl_program.cpp:140: GL program(:null) link failed, error flag: 0x0000:GL_NO_ERROR, link log:error: Too many compute shader storage blocks (13/12)

XCAM ERROR gl_program.cpp:202: GLProgram(null) link program failed
XCAM ERROR gl_image_shader.cpp:181: GLImageShader(shader_reconstruct_pyr_yuv420) program(null) pour shader failed
XCAM ERROR gl_blender.cpp:225: gl-blender create shader_reconstruct_pyr_yuv420 program failed
test-surround-view: gl_blender.cpp:383: XCamReturn XCam::GLBlenderPriv::BlenderImpl::create_shaders(): Assertion `layer.reconstruct.ptr ()' failed.
Aborted (core dumped)

Douaty, 2023-06-09:
> Happens when built without `--enable-gles`.

Hi， May I ask the following questions? Thanks
./test-surround-view --module gles --input test/01.nv12 --input test/02.nv12 --output test/output.yuv --in-w 640 --in-h 360 --out-w 1120 --out-h 480 --topview-w 640 --topview-h 360 --in-format yuv --fisheye-num 2 --cam-model cama2c1080p --blend-pyr-levels 2 --dewarp-mode sphere --scopic-mode mono --scale-mode dualconst --frame-mode single --loop 1
input0 file:		test/01.nv12
input1 file:		test/02.nv12
camera model:		cama2c1080p
fisheye number:		2
stitch module:		GLES
enable DMA buffer input/output:		false
device node:		Not specified, use default model
output file:		test/output.yuv
input width:		640
input height:		360
output width:		1120
output height:		480
topview width:		640
topview height:		360
cubemap width:		1280
cubemap height:		720
input format:		yuv
blend pyr levels:	2
dewarp mode: 		sphere
scopic mode:		mono
scaling mode:		dualconst
feature match:		none
frame mode:		singleframe
save output:		true
save topview:		false
save cubemap:		false
loop count:		1
repeat count:		1
XCAM ERROR gl_program.cpp:140: GL program(:null) link failed, error flag: 0x0000:GL_NO_ERROR, link log:error: Too many compute shader storage blocks (13/12)

XCAM ERROR gl_program.cpp:202: GLProgram(null) link program failed
XCAM ERROR gl_image_shader.cpp:181: GLImageShader(shader_reconstruct_pyr_yuv420) program(null) pour shader failed
XCAM ERROR gl_blender.cpp:225: gl-blender create shader_reconstruct_pyr_yuv420 program failed
test-surround-view: gl_blender.cpp:383: XCamReturn XCam::GLBlenderPriv::BlenderImpl::create_shaders(): Assertion `layer.reconstruct.ptr ()' failed.
Aborted (core dumped)

zongwave, 2023-06-12:
Thanks for report issues, submitted patch to fix this issue.
https://github.com/intel/libxcam/pull/827