intel/libxcam #825: test-surround-view: command not found

lekhashreehj, 2023-05-23:
When I am trying run the following code I am getting this error
test-surround-view --module soft --input input.nv12 --output output.nv12 --in-w 1920 --in-h 1080 --out-w 1920 --out-h 960 --in-format nv12 --fisheye-num 2 --res-mode 1080p2cams --blend-pyr-levels 1 --dewarp-mode sphere --scopic-mode mono --scale-mode dualconst --fm-mode capi --fm-frames 120 --fm-status fmfirst --frame-mode multi --save true --save-topview false --loop 1

Please help me to debug this error

zongwave, 2023-05-23:
can you post the error message?

lekhashreehj, 2023-05-23:
test-surround-view: command not found
Its showing command line not found.
I am trying to build it with CPU and not using any hardware. 
I am stuck in this step.

shreshtashetty, 2023-05-23:
Try 
`cd tests`
And then run the whole command you posted in your question starting with `./`

lekhashreehj, 2023-05-23:
Thank you its working!!!

zongwave, 2023-05-24:
Thanks @shreshtashetty

closed 2023-05-24 by zongwave